OrcaSlicer/OrcaSlicer #6454: Scaling option doesn't work in Orca Slicer CLI

uSyzex, 2024-08-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

i7 6700k, 32GB RAM, 3070Ti

### Printer

Bambu X1C

### How to reproduce

1. `./orca-slicer.exe --scale 2 --slice 1 --allow-newer-file test_model.stl --debug 5`
OR 
2. `./orca-slicer.exe --scale 2 --info --allow-newer-file test_model.stl --debug 5`

It doesn't work either with STLs or 3mf

### Actual results

Error in the trace (debug 5) and nothing happens when no debug level is enabled. 

### Expected results

Scale the model and execute the action, info or slice. 

### Project file & Debug log uploads

```
[2024-08-13 17:32:06.686301] [0x0000d264] [trace]   Initializing StaticPrintConfigs
[2024-08-13 17:32:06.701300] [0x0000d264] [warning] cli mode, Current OrcaSlicer Version 01.09.03.50
[2024-08-13 17:32:06.701300] [0x0000d264] [info]    Will start to read model file now, file count :1

[2024-08-13 17:32:06.702301] [0x0000d264] [info]    allow_multicolor_oneplate 1, allow_rotations 1 skip_modified_gcodes 0 avoid_extrusion_cali_region 0 loaded_filament_ids size 0, clone_objects size 0, skip_useless_pick 0, allow_newer_file 1
[2024-08-13 17:32:06.702301] [0x0000d264] [info]    plate_to_slice=1, normative_check=1, use_first_fila_as_default=0
[2024-08-13 17:32:06.702301] [0x0000d264] [info]    read model file:test_model.stl

[2024-08-13 17:32:06.758041] [0x0000d264] [debug]   TriangleMesh::repair() started
[2024-08-13 17:32:06.760042] [0x0000d264] [debug]   TriangleMesh::repair() finished
[2024-08-13 17:32:06.764040] [0x0000d264] [info]    object test_model.stl, id :5, from stl or other 3mf

[2024-08-13 17:32:06.764040] [0x0000d264] [warning] 1632: not support to slice plate 1, reset to 0
[2024-08-13 17:32:06.764040] [0x0000d264] [info]    Slic3r::CLI::run:before load settings, file count=0

[2024-08-13 17:32:06.764040] [0x0000d264] [info]    current printer , new printer , current process , new process
[2024-08-13 17:32:06.764040] [0x0000d264] [info]    current printer inherits , new printer inherits , current process inherits , new process inherits
[2024-08-13 17:32:06.764040] [0x0000d264] [info]    old 3mf, no compatible printers, set to compatible
[2024-08-13 17:32:06.764040] [0x0000d264] [info]    old printer , inherited from , old process , inherited from  ,compatible 1
[2024-08-13 17:32:06.764040] [0x0000d264] [warning] no filament colors found in projects
[2024-08-13 17:32:06.764040] [0x0000d264] [info]    total 1 models, 0 objects

[2024-08-13 17:32:06.765041] [0x0000d264] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from 1517 to 0
[2024-08-13 17:32:06.765041] [0x0000d264] [debug]   Slic3r::GUI::PartPlate::set_index: plate_id update from 0 to 1
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    translate_old 0, shrink_to_new_bed 0, old bed size {200, 200, 100}
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlateList::reset_size:before size: plate_width 0, plate_depth 0, plate_height 0
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlateList::reset_size:after size: plate_width 200, plate_depth 200, plate_height 100
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: plate_id 0, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: with_instance_move 0, after, origin {0,0,0}, plate_width 200, plate_depth 200, plate_height 100
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: plate_id 1, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlate::set_pos_and_size: with_instance_move 0, after, origin {240,0,0}, plate_width 200, plate_depth 200, plate_height 100
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-13 17:32:06.765041] [0x0000d264] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-13 17:32:06.765041] [0x0000d264] [error]   Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles

[2024-08-13 17:32:06.765041] [0x0000d264] [info]    will start transforms, commands count 1

[2024-08-13 17:32:06.765041] [0x0000d264] [info]    process transform scale

[2024-08-13 17:32:08.101907] [0x0000d264] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-13 17:32:08.101907] [0x0000d264] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2024-08-13 17:32:08.102909] [0x0000d264] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2024-08-13 17:32:08.102909] [0x0000d264] [info]    Slic3r::GUI::PartPlate::update_slice_result_valid_state: plate 1 , update slice result from 0 to 0

```

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I have tried multiple way to format the scale option, none of them worked so I think it's more of a Slicer problem than just a CLI args not being formatted correctly. 

I have opened two issues on Bambu-Studio repo regarding CLI issues:

- https://github.com/bambulab/BambuStudio/issues/4628
- https://github.com/bambulab/BambuStudio/issues/4627

NiXTheDev, 2024-08-13:
Almost the same error, but only that it doesn't output anything at all, even on `--debug 5`

uSyzex, 2024-08-13:
> Almost the same error, but only that it doesn't output anything at all, even on `--debug 5`

Can you please provide your log, it could help fixing it.

uSyzex, 2024-08-19:
Bug found in Bambu Studio. https://github.com/bambulab/BambuStudio/issues/4628#issuecomment-2296050127

NiXTheDev, 2024-08-24:
> > Almost the same error, but only that it doesn't output anything at all, even on `--debug 5`
> 
> Can you please provide your log, it could help fixing it. 

There is no log to send, it doesn't output nothing

uSyzex, 2024-08-26:
A fix has been implemented in BambuStudio, anyway to push it to Orca ? 
https://github.com/bambulab/BambuStudio/commit/2d4655e780e85e84445bb0b1987445e9e8d66b95

uSyzex, 2024-08-30:
https://github.com/bambulab/BambuStudio/issues/4628#issuecomment-2320387581

uSyzex, 2024-09-29:
It's been a month it has been pushed to BS, is there any ETA ?
Regards

PR #6969 (2024-10-07): FIX: CLI: fix an exception when parsing scale params
# Description

This PR aims to solve an exception when using OrcaSlicer in CLI which caused the --scale option to not work. 
All credits to @lanewei120 from BambuStudio for pushing it on BS repo.

Related commit on BS:
https://github.com/bambulab/BambuStudio/commit/2d4655e780e85e84445bb0b1987445e9e8d66b95#diff-2fa194122e892282ec589e51d5f7da65da83ec0488ced4b63930345d1273a353R3797
Related issue on BS:
https://github.com/bambulab/BambuStudio/issues/4628
Related issue on Orca:
https://github.com/SoftFever/OrcaSlicer/issues/6454
fixes #6454
## Te [... cut]
commit 9fbab140 (2024-10-07): FIX: CLI: fix an exception when parsing scale params (#6969)

# Description

This PR aims to solve an exception when using OrcaSlicer in CLI which
caused the --scale option to not work.
All credits to @lanewei120 from BambuStudio for pushing it on BS repo.

Related commit on BS:

https://github.com/bambulab/BambuStudio/commit/2d4655e780e85e84445bb0b1987445e9e8d66b95#diff-2fa194122e892282ec589e51d5f7da65da83ec0488ced4b63930345d1273a353R3797
Related issue on BS:
https://github.com/bambulab/BambuStudio/issues/4628
Related issue on Orca:
https://github.com/SoftFever/OrcaSlicer/issues/6454
fixes #6 [... cut]
closed 2024-10-07 by SoftFever