OrcaSlicer/OrcaSlicer #6458: Fan speed on layer change: SET_FAN_SPEED_CHANGING_LAYER

mbiggs77, 2024-08-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I see that gcode includes the comment ;_SET_FAN_SPEED_CHANGING_LAYER on every layer change, but I can't find anywhere to add the setting in the software.
I am working on a part with ASA where the main body of the print works great with one setting, but it ends with a thinner piece that doesn't cool enough and therefore doesn't print properly.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be great to add a parameter for fan speed as an editable value for different layer ranges.

### Describe alternatives you've considered

By manually adding a M105 S51 line in the gcode it printed perfectly, but it would be ideal to not have to edit the gcode to do this.

### Additional context

This issue has been discussed before but closed.

discip, 2024-08-14:
@mbiggs77
You can:
1. move the slider to the desired height
2. click the plus on the slider with the right mouse button
3. and choose **`Add Custom G-code`**
4. enter `M105 S51` in the now opened dialog-box.

discip, 2024-08-14:
@mbiggs77
You can:
1. move the slider to the desired height,
2. click the plus on the slider with the right mouse button,
3. choose **`Add Custom G-code`**
4. and enter `M105 S51` in the now opened dialog-box.

mbiggs77, 2024-08-14:
That's fabulous! Thank you @discip!

discip, 2024-08-20:
@mbiggs77
As you seem to be happy, please don't forget to close this issue. 😊

closed 2024-08-20 by mbiggs77

BigToyBox, 2025-04-19:
I think you meant M106 S51 and I am surprised no one has commented that M106 controls the fan speed