OrcaSlicer/OrcaSlicer #6459: strange "asymmetric" print quality, print settings ignored on some parts of multi-object print job? (Pressure advance, others?)

sygolene, 2024-08-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD Ryzen 7 3700X
32 GB ram
RTX 3060 TI

### Printer

Bambu Lab P1S, firmware 01.06.01.00

### How to reproduce

1.import model into Orca project
2.arrange the two parts orthogonally (see attached project file)
3.use proper custom profile for filament (self made Fiberlogy Matte PETG)
4.print

### Actual results

the two halves aren't applying the same printing settings as shown on the attached jpegs.
[image]
[image]
[image]

### Expected results

As one can see on the photos, one part has a smooth surface and proper corners while the other show bulged corners & banding. Both part shared the same orientation (long side on x axis). 
I'm mostly interested in the PA issue but am puzzled by the banding. I know this is typically related to the printer itself (speed, axis, ...) but here I'm puzzled to see that one part is unaffected by the banding so I'm guessing other settings are applied when printing the layer of object "bad".
Individual object settings were left unaltered and are identical. 
I'm considering this as a bug but I might be wrong, any help appreciated. 

### Project file & Debug log uploads

[Github-Issue - Clean - refactor - clipped.zip](https://github.com/user-attachments/files/16611050/Github-Issue.-.Clean.-.refactor.-.clipped.zip)
[debug_Sun_Aug_11_19_38_56_2160.log.zip](https://github.com/user-attachments/files/16611171/debug_Sun_Aug_11_19_38_56_2160.log.zip)
not entirely sure about the log file but it was my only print that day so should be ok...

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2024-08-15:
This is purely printer mechanics. The bambu X and P series has a known issue with the carbon rods being "sticky" when moving on the X axis. This causes VFA's (the ripples that you see) especially towards the rear of the build chamber. It cannot be solved (much less in the slicer) as its inherent to the design and material choices for the gantry.

What you can do, is make sure the carbon rods are squeaky clean using IPA plus make sure the gantry is squared by adjusting the belt tension. Finally print faster or print slower. The artefacts appear at a specific speed range between approx 70/80mm/sec and 160-200mm/sec and are worse around 120mm/sec.

sygolene, 2024-08-15:
Hello, thanks for your reply.
VFA is not the important thing here, as I mentioned in my post : I'm more
puzzled by the difference in finish on same axis surfaces. VFA is one such
manifestation.
More notably the corner finish which seem to indicate PA isn't applied in
multi part things?

Thanks,

kylek29, 2024-08-15:
One way to test this further is to open the same file and then swap the object placements but change nothing else (so where the good object was, place the other object) .. literally just drag/drop. 

That'll tell you if it's likely to be related to the machine (e.g. the formerly good object becomes the bad object). Another thing (might not be applicable here) is that distance to the Aux Fan (if you have it and it's on) can affect things as well. The default Aux tends to blow across the center.

igiannakas, 2024-08-15:
Variances in corner appearance due to PA are impossible to be created by the slicer - PA is set at the beginning of the print (and unless you've enabled adaptive PA in 2.2 Beta) and it stays the same irrespective of the item on the build plate (it is not set again by the slicer).

The variance in corner performance could be because of mechanical issue or calibration (too low PA) or **just the physical limits of printing in FFF.**

For example you'll always see in a corner, **before the corner there is a slight rounding** and **after the corner a slight bulge**. That is also because of the way the toolhead moves. **Going into the corner, it "cuts" the corner** a bit because it can't change direction instantly, so it appears rounded. Same for **going out of a corner - it bulges a bit** because it can't change direction instantly. So if one part is showing the "going into a corner" side and the other part the "coming out of a corner" side, you'll get this visual difference. 

A way to reduce the influence of that lack of instant direction change is to reduce your print speed, jerk and acceleration. Thatway you're allowing more time to the toolhead to do the direction change.

igiannakas, 2024-08-15:
For example, in your project file the parts on the front and rear of the plate are mirrored. 

So you are indeed seeing the going in a corner / coming out of a corner variance mentioned above:

[image]

igiannakas, 2024-08-15:
What you can do to keep the “going in a corner” rounding facing the same side if you want, is to change the print direction of the perimeters an object - under the quality tab in the object settings you can select the print direction to be clockwise, counter clockwise wise and auto. Pick clockwise for one of the two objects.

igiannakas, 2024-08-15:
[image]

sygolene, 2024-08-15:
Thanks very much, I better understand the issue. And thanks for the setting tips, I'll give it a try for my next run.

closed 2024-08-15 by sygolene