OrcaSlicer/OrcaSlicer #6464: Independently adjust X / Y compensation

TheVoronModder, 2024-08-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I noticed slicers do not have X / Y compensation for those "good-enough-ists" out there, with Core XY for example, it is crucial to have good belt tension, not everybody is willing to spend the time adjusting belts, so, simple solution, print a 30mm cube then adjust the X / Y in your slicer.

Say you need 30mm on x but you are getting 30.05, you would compensate by doing X = -.05

same for Y, say it is 30.15 and you want 30, you would compensate for Y = -.15

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I want to be able to tell my slicer how much to compensate for X Y and Z axis instead of relying on Rotational Distances, I believe adjusting the slicer based on what material you are using, IE: PLA, ASA bla bla, you should be able to have dimensional compensation.

Sure, you could just re-size your part subtracting or adding the distances you need however this does not really work well with circles. 

allow us to compensate with X and Y or one or the other.

### Describe alternatives you've considered

Rotational Distance Calculations (this is EVIL according to Voron Purists)

Flow Compensation

manually adjusting belts

### Additional context

The above should be self explanatory, if not please message me!

avogra, 2024-09-18:
Just as a hint regarding calibration cubes: Typically they are not the right tool to determine dimensional accuracy. See this video for explanation: https://youtu.be/H7OsnMLDIMw?si=c71UnzCDs_32r8MR

I'm not sure if you are aware of the shrinkage compensation, which allows you to at least compensate X and Y together for each filament. Typically, there is no need to compensate Z, if your printer mechanics are setup correctly.

You can find "shrinkage compensation" in material settings, tab: filament, in the section: "basic information".

TheVoronModder, 2024-10-02:
allow me to explain better:

Sure, shrinkage comp works well for X and Y TOGETHER.

Im asking for an option that allows a user to compensate X and or Y NOT together.

If my Y is always dimensionally accurate and my X is off a bit, I would like to adjust my slicer settings to compensate for the printer.

This is NOT for me, just thought it would be a neat feature for "good enoughist" out there who want more accuracy out of their 3d printers using Orca Slicer.

avogra, 2024-10-05:
Thank you for the clarification. I saw your mention of rotational distances (which I didn't understand) in contrast to linear scaling in X/Y. Now it is clear that you want the existing compensation of Orca to be split into separate X and Y values.

closed 2025-01-11 by github-actions[bot]