OrcaSlicer/OrcaSlicer #6465: Brim material same as object outer wall

Dynom, 2024-08-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When adding brim to an object with multiple colours on the same first layer, the brim could be printed with the same material as the first colour, which isn't necessarily the outer most wall.
This could result in color bleeding at the edges, especially when the colours contrast greatly (e.g. black and white).

Normally when removing the brim in the same colour the effort is rather painless, but if there is a little bleeding, the post-processing can be quite a bit of work. Since there is no easy way to change the brim colour now, I think it would make for a lovely addition.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Automatically (or manually) select the same material for the brim that's also used for the outer walls.

discip, 2024-08-16:
@Dynom
Try the following:

[image]

In step 4 you need to rearrange the colors by dragging the squares.

Maybe this could solve what you are asking for.

Dynom, 2024-08-17:
Thanks, I'll try that as a work-around, however a simple option would have my preference, to prevent these bleeds:

[image]

Steiner-SE, 2025-04-20:
Not solved, no matter how you rearrange color priority it will still always prime using color 1, even if that isn't used at all in the print, this is causing excessive and instant filament changes.

jimothyr, 2026-01-17:
As an extension of this issue, when printing by object if the second object has a new first layer colour, the slicer still uses the first colour as the brim. This leads to unnecessary colour changes. with each object getting the first objects brim colour. 

[image]