OrcaSlicer/OrcaSlicer #6466: Bambu Virtual Camera Tools installation loop

kevireilly, 2024-08-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5

### Additional system information

Apple M1 Max

### Printer

Bambu Labs X1C

### How to reproduce

1. Go to the Device tab
2. Click the Camera settings
3. Click Go Live switch
4. Click OK to install Virtual Camera Tools

### Actual results

When Virtual Camera Tools reaches 50%, the dialog to install Virtual Camera Tools is invoked again. This cycle appears to repeat indefinitely.
[image]

### Expected results

Virtual Camera Tools are successfully installed without entering an installation loop. It is worth noting that on Bambu Studio, the download/installation process also only reaches 50%, but it does succeed and Go Live is enabled appropriately.

### Project file & Debug log uploads

[debug_Thu_Aug_15_00_17_04_56362.log.0.zip](https://github.com/user-attachments/files/16622569/debug_Thu_Aug_15_00_17_04_56362.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

shiggitay, 2024-08-22:
I can confirm that this behavior happens even on v2.1.0 (the release before most current).

pkilar, 2024-09-03:
The same bahavior on Linux. tried both v2.1.1 and latest nightly build.

lightmaster, 2024-09-04:
Slightly different issue on Windows. It gets to 40% downloaded and then says "Failed to install the plug-in. Please check whether it is blocked or has been detected by anti-virus software." Then includes a link to https://wiki.bambulab.com/en/software/bambu-studio/failed-to-get-network-plugin

yenba, 2024-09-13:
Just to add another, this also happens to me on macOS 14.6.1 running the latest stable version of Orca v2.1.1.

rudolfp1979, 2024-09-16:
Same for me BambuStudio no problem in bambu studio. But to clear with 2.0 it installs it after upgrading to 2.2.0 all works till reboot then it only works when bambustudio turns on virtualcamera even if it takes the file from orcaslicer directory. Could it be that virtual camera server is not starting?

SRFUKB, 2024-09-16:
I had this on the nightly using windows and then 2.2.0. I removed all instances of Orca from the firewall and then ran the portable version, this time the firewall prompt came up and I approved and it installed the plugin OK and now works.

lightmaster, 2024-09-16:
> I had this on the nightly using windows and then 2.2.0. I removed all instances of Orca from the firewall and then ran the portable version, this time the firewall prompt came up and I approved and it installed the plugin OK and now works.

So the plugin is working in the portable verison, or in the normal install version?

rudolfp1979, 2024-09-17:
Me is using the Mac arm Version the opener of this case uses this one too.

SRFUKB, 2024-09-17:
> So the plugin is working in the portable verison, or in the normal install version?

Sorry for the delayed reply.

It is the portable, I never use the normal install as I can run the newer beta version and always go back if needed.

I had about 20 instances of Orca in the firewall so the clean up was long due.

I thought it was an issue with the FW because it got to 40/50% and at that point I would get a FW prompt to allow it through the FW. In this case it didnt come up. I do think a Bug exists, its just I was able to get around it.

pegoto, 2024-11-08:
Have this problem too, and for a while on nightly, beta and prod version.

altjx, 2024-12-21:
Issue is still persistent 4 months later in December. Is this on any list to get fixed?

SRFUKB, 2024-12-21:
I still use the portable version and have had no repeat issue in 2.2.0 or the nightly 2.3.0 builds. I use windows 12 latest updates on a intel i7 10700. I checked the nightly released yesterday and no issue with the plugin at all. This all resolved when I cleaned out firewall rules for the older versions of Orca as to which networks it could use (Private or Public).

rudolfp1979, 2024-12-21:
I think all people here use the normal Version

rudolfp1979, 2024-12-21:
There are no Mac portables

GCheung55, 2024-12-28:
I'm seeing this issue on a macOS Sonoma v14.5, Intel i9, running Orca Slicer v 2.2.0. Virtual camera tools installation stops at 50% and reopens dialog to install tools.

gloriouslyawkwardlife, 2025-01-15:
Same issue here. MacBook Air M3 13”, Sequoia 15.2, OrcaSlicer 2.2.0.

lexworth, 2025-01-19:
This is still an issue for me on iMac with OrcaSlicer 2.2.0. I tried copying the cameratools folder from the Bambustudio install but it still would not allow livestream to OBS.

vpassapera, 2025-03-20:
Still happening on linux 2.2.0 using appimage version.

tk-bl, 2025-04-20:
Still an issue.  Ubuntu 24.04, Orca Slicer 2.3.0.  Gets to 50%, then goes back to the window prompting to install the virtual camera tools.

tkeifer, 2025-07-17:
Running into this on Sequoia (15.5) + Orca v2.3.0

DeepReef11, 2025-07-18:
Does it work for anyone? Same problem, linux, Orca v2.3.0

DC4JG, 2025-08-03:
Can reproduce the issue.

MacOS: Version 15.5 (24F74)
Orca: [Version 2.3.0 Mac arm64](https://github.com/SoftFever/OrcaSlicer/releases/download/v2.3.0/OrcaSlicer_Mac_arm64_V2.3.0.dmg)

raistlin7447, 2026-09-05:
I think I found the cause of the install loop.

Fix is in #15540. I don't have a Bambu printer, so I can't test it. If someone hitting this can try a build from that PR I would appreciate it. The artifact link is on the PR under "How to Download Pull Requests Artifacts for Testing". You'll need a github account to be able to see the artifacts.

If you are already logged in, you can go straight here and scroll down to the bottom to get your download: https://github.com/OrcaSlicer/OrcaSlicer/actions/runs/33929499263

Please let me know what Operating System you test in.

zamro, 2026-09-06:
Kubuntu 24.04 here.
Just started using Bambu X2D and Orca Slicer today and immediately stumped on this problem. 
OrcaSlicer-Linux-flatpak_PR-15540_x86_64.flatpak worked for me, after installing it, cameratools downloaded and installed in the first try.

Thank you very much for finding the underlying issue!

closed 2026-09-06 by RF47