OrcaSlicer/OrcaSlicer #6467: "Apply Gap Fill" Setting Not Functioning as Expected

TobiasHeimGalindo, 2024-08-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

BambuLab X1C

### How to reproduce

1. Open OrcaSlicer.
2. Load any 3D model into the slicer.
3. Go to the "Apply gap fill" setting, which is set to "Nowhere" by default.
4. Slice the model and observe that gap fill is still being applied universally.

### Actual results

[image]

### Expected results

Workaround for Applying Gap Fill nowhere:

1. Change the "Apply gap fill" setting to "Everywhere."
2. Set the "Filter out tiny gaps" parameter to an extremely high value (e.g., 1000mm).
3. Re-slice the model, and notice that gap fill is now correctly excluded.

[image]

### Project file & Debug log uploads

Due to the nature of the project file (as it is a commercial product), I am unable to upload it. However, the issue is easily reproducible using any model by following the steps outlined above.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-08-15:
**There are two types of gap fill:**
1. Fill small gaps over solid infill
2. Fill gaps between perimeters when using the classic perimeter generator.

This setting controls the former (1) -> i.e. it applies gap fill to the top, bottom, infill or none of those areas

The gap fill between perimeters is not controlled by this setting (hence why this setting is in the infill category, not the perimeters section). You should not and must not disable gap fill between perimeters as you are in effect removing strength from your model.

If you really know what you're doing there is a "workaround" to fully disable gap fill - set the gap fill threshold to 99999 and you'll remove the gap fill between perimeters. However that is not advised unless you really have a specific case for it. And most likely if you do, Arachne slicing would be a better way to go.

TobiasHeimGalindo, 2024-08-15:
> **There are two types of gap fill:**
>
>     1. Fill small gaps over solid infill
>
>     2. Fill gaps between perimeters when using the classic perimeter generator.
>
>
> This setting controls the former (1) [...]

Wow that was great advice! I didn't know about the Arachne slicing, for my use case this is much better. Much appreciated!

igiannakas, 2024-08-15:
No problem! I've pushed a PR clarifying that tooltip and giving the same advice as above.

**Tooltip:** 
Enables gap fill for the selected solid surfaces. The minimum gap length that will be filled can be controlled from the filter out tiny gaps option below.

Options:

Everywhere: Applies gap fill to top, bottom and internal solid surfaces for maximum strength
Top and Bottom surfaces: Applies gap fill to top and bottom surfaces only, balancing print speed, reducing potential over extrusion in the solid infill and making sure the top and bottom surfaces have no pin hole gaps
Nowhere: Disables gap fill for all solid infill areas.
Note that if using the classic perimeter generator, gap fill may also be generated between perimeters, if a full width line cannot fit between them. That perimeter gap fill is not controlled by this setting.

If you would like all gap fill, including the classic perimeter generated one, removed, set the filter out tiny gaps value to a large number, like 999999.

However this is not advised, as gap fill between perimeters is contributing to the model's strength. For models where excessive gap fill is generated between perimeters, a better option would be to switch to the arachne wall generator and use this option to control whether the cosmetic top and bottom surface gap fill is generated.

discip, 2024-08-16:
@TobiasHeimGalindo
Please don't forget to close as this seems to be solved. 😊

igiannakas, 2024-08-16:
No worries it will auto close when the Pr is merged ;)

PR #6468 (2024-08-18): Gap fill tool tip clarification & filter_out_gap_fill setting visibility
# Description

Clarified the gap fill selection drop down tool tip.

fixes: #6467 

**Edit:**
Following feedback, the tooltip of filter_out_gap_fill is also amended for clarity as well as the setting not being hidden any more if None is selected on the apply gap fill option, as this also controls the classic wall generator gap fill filter.
closed 2024-08-16 by TobiasHeimGalindo