OrcaSlicer/OrcaSlicer #6470: Support Filament Painting / Support Modifiers

mpetito, 2024-08-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

While printing some parts that require support, the need for a special support interface filament might vary for different parts of a single object. Many sections may print fine without a support interface material while other sections may demand a support interface material.

For example I may have a long flat bridge where I do not want any scarring on the underside from support, and so I want to use interface material there, while elsewhere I may have areas that need some support but a standard interface will work just fine.

I certainly could use the support filament _everywhere_, but this increases print costs and especially time for material swaps.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to control the support filament used while painting supports onto an object or to vary certain support process settings through modifiers.

Support settings generally vary significantly with / without support filament (e.g. top z distance of 0).  While I think it would be more convenient to select the support filament while painting, I can understand how that might not easily map to the correct support interface settings for a single object.

If a modifier approach is used, then the painting tool remains unchanged but the available modifier options should minimally include:
* Top / Bottom Z Distance
* Top / Bottom Interface Layers
* Top / Bottom Interface Spacing
* Interface Pattern

### Describe alternatives you've considered

1. Modifiers - Modifiers cannot change the relevant support settings.
2. Split to Parts - Distinct parts within an assembly cannot have varying support settings.
3. Floating Assemblies - Objects can have varying support settings, so I can align the "parts as objects" by creating aligned floating assemblies (using a grounding cube), and then modify the support settings of these assemblies. This only works in simple scenarios since supports may generate through other assemblies.

### Additional context

[image]

In this example I have split the upper and lower areas into separate assemblies so that I can use a support interface filament (yellow) in the lower region only.

For this object:
* Printing with support filament applied throughout requires 88 filament changes and takes 10hrs.
* Printing with support filament selectively applied requires 4 filament changes and takes 5hrs.

Note this exercise may have also surfaced a bug where support filament is still flushed into the support of the upper assembly even though I have "Flush into objects' support" disabled globally and for both assemblies.

sconeboi666, 2024-08-27:
I's also agree to why this isnt a feature.  this would have been perfect for me when decided where id like to use support material other than the Default. this would also eliminate waste as well

Daubydoo, 2024-08-28:
The ability to change support settings for modifiers would be a game changer. I have a part that requires a massive amount of support and I would like to be able to set the "Base Pattern Spacing" to 15-20mm for most of the support and then adaptively change the density (3-5mm) as it gets closer to the interface layers with support modifiers, this would save a lot of time and material. 
[image]

mpetito, 2024-08-28:
@Daubydoo I suspect that using modifiers to change the support structure vs the support interface may best be represented in separate issues.

Looking through the code a bit, each of the support types has its own code to generate the structure, whereas the interface layers appear to be generated by some other code which is independent of the support type.

sconeboi666, 2024-08-28:
I agree as per above. I was happy with the supports themself, but would like to have changed the support interface per modifier

closed 2024-12-04 by github-actions[bot]