OrcaSlicer/OrcaSlicer #6471: OrcaSlicer Crashes on Startup Arch Linux

aeonik, 2024-08-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1 (AppImg)

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Additional system information

```
❯ sudo inxi -Faz
System:
  Kernel: 6.10.4-arch2-1 arch: x86_64 bits: 64 compiler: gcc v: 14.2.1
    clocksource: tsc avail: hpet,acpi_pm parameters: initrd=\amd-ucode.img
    initrd=\initramfs-linux.img
    root=PARTUUID=fa3f6bd1-daf4-4db7-b041-680a4658ef18 acpi_osi=Linux
    crashkernel=768M audit_backlog_limit=8192 acpi_enforce_resources=lax
    sysrq_always_enabled=1 nvme_core.default_ps_max_latency_us=5500
    amdgpu.ppfeaturemask=0xffffffff rw
  Desktop: KDE Plasma v: 6.1.4 tk: Qt v: N/A wm: kwin_x11 with: krunner
    tools: avail: xsecurelock,xtrlock dm: SDDM Distro: Arch Linux
Machine:
  Type: Desktop Mobo: Micro-Star model: MEG X570 GODLIKE (MS-7C34) v: 1.0
    serial: <filter> uuid: a5b8a094-208a-b71e-a414-2cf05d7a3960 UEFI: American
    Megatrends LLC. v: 1.L0 date: 06/30/2023
CPU:
  Info: model: AMD Ryzen 9 5950X socket: AM4 bits: 64 type: MT MCP
    arch: Zen 3+ gen: 4 level: v3 note: check built: 2022 process: TSMC n6 (7nm)
    family: 0x19 (25) model-id: 0x21 (33) stepping: 0 microcode: 0xA201025
  Topology: cpus: 1x cores: 16 tpc: 2 threads: 32 smt: enabled cache:
    L1: 1024 KiB desc: d-16x32 KiB; i-16x32 KiB L2: 8 MiB desc: 16x512 KiB
    L3: 64 MiB desc: 2x32 MiB
  Speed (MHz): avg: 2350 high: 3400 min/max: 2200/5083 boost: enabled
    base/boost: 3400/5050 scaling: driver: acpi-cpufreq governor: schedutil
    volts: 1.1 V ext-clock: 100 MHz cores: 1: 2200 2: 2200 3: 3400 4: 2200
    5: 2200 6: 2200 7: 2200 8: 2200 9: 2200 10: 2200 11: 2200 12: 2200
    13: 2200 14: 2200 15: 2200 16: 2200 17: 3400 18: 2200 19: 3400 20: 2200
    21: 2200 22: 2200 23: 2200 24: 3400 25: 2200 26: 2200 27: 2200 28: 2200
    29: 2200 30: 2200 31: 2200 32: 2200 bogomips: 217674
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
  Vulnerabilities:
  Type: gather_data_sampling status: Not affected
  Type: itlb_multihit status: Not affected
  Type: l1tf status: Not affected
  Type: mds status: Not affected
  Type: meltdown status: Not affected
  Type: mmio_stale_data status: Not affected
  Type: reg_file_data_sampling status: Not affected
  Type: retbleed status: Not affected
  Type: spec_rstack_overflow status: Vulnerable: Safe RET, no microcode
  Type: spec_store_bypass mitigation: Speculative Store Bypass disabled via
    prctl
  Type: spectre_v1 mitigation: usercopy/swapgs barriers and __user pointer
    sanitization
  Type: spectre_v2 mitigation: Retpolines; IBPB: conditional; IBRS_FW;
    STIBP: always-on; RSB filling; PBRSB-eIBRS: Not affected; BHI: Not
    affected
  Type: srbds status: Not affected
  Type: tsx_async_abort status: Not affected
Graphics:
  Device-1: AMD Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] vendor: Tul /
[... 528 lines cut ...]
        libdav1d.so.7 => /usr/lib/libdav1d.so.7 (0x00007f2ee697b000)
        librav1e.so.0.7 => /usr/lib/librav1e.so.0.7 (0x00007f2ee6600000)
        libSvtAv1Enc.so.2 => /usr/lib/libSvtAv1Enc.so.2 (0x00007f2edde00000)
        libaom.so.3 => /usr/lib/libaom.so.3 (0x00007f2edd400000)
        libpsl.so.5 => /usr/lib/libpsl.so.5 (0x00007f2ee8976000)
        libgssapi_krb5.so.2 => /usr/lib/libgssapi_krb5.so.2 (0x00007f2ee6928000)
        libtss2-esys.so.0 => /usr/lib/libtss2-esys.so.0 (0x00007f2ee6563000)
        libtss2-mu.so.0 => /usr/lib/libtss2-mu.so.0 (0x00007f2ee6513000)
        libtss2-rc.so.0 => /usr/lib/libtss2-rc.so.0 (0x00007f2ee896a000)
        libtss2-tctildr.so.0 => /usr/lib/libtss2-tctildr.so.0 (0x00007f2ee8960000)
        libevdev.so.2 => /usr/lib/libevdev.so.2 (0x00007f2ee64fe000)
        libxcb-randr.so.0 => /usr/lib/libxcb-randr.so.0 (0x00007f2ee64ec000)
        libbrotlicommon.so.1 => /usr/lib/libbrotlicommon.so.1 (0x00007f2ee64c9000)
        libudev.so.1 => /usr/lib/libudev.so.1 (0x00007f2ee6483000)
        libunistring.so.5 => /usr/lib/libunistring.so.5 (0x00007f2edd250000)
        libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007f2ee6461000)
        libkrb5.so.3 => /usr/lib/libkrb5.so.3 (0x00007f2eddd3b000)
        libk5crypto.so.3 => /usr/lib/libk5crypto.so.3 (0x00007f2eddd0e000)
        libcom_err.so.2 => /usr/lib/libcom_err.so.2 (0x00007f2ee8956000)
        libkrb5support.so.0 => /usr/lib/libkrb5support.so.0 (0x00007f2ee6453000)
        libkeyutils.so.1 => /usr/lib/libkeyutils.so.1 (0x00007f2ee6921000)
        libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007f2ee6441000)
        libcrypto.so.3 => /usr/lib/libcrypto.so.3 (0x00007f2edcc00000)
        libtss2-sys.so.1 => /usr/lib/libtss2-sys.so.1 (0x00007f2eddce7000)
```

lzhecz, 2024-08-15:
Same Issue, bambu-studio crashes too

```
[lzhecz@arch ~]$ orca-slicer
[2024-08-15 21:40:49.050649] [0x0000700b1b8d8b80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
zsh: segmentation fault (core dumped)  orca-slicer
```

faith-healer, 2024-08-15:
same....
> [2024-08-15 22:05:46.193722] [0x00007cf6df338b80] [trace]   Initializing StaticPrintConfigs
> add font of HarmonyOS_Sans_SC_Bold returns 1
> add font of HarmonyOS_Sans_SC_Regular returns 1
> add font of NanumGothic-Regular returns 1
> add font of NanumGothic-Bold returns 1
> Segmentation fault (core dumped)

Sanaki, 2024-08-15:
Same experience here on Garuda Linux.

Related bambustudio issue: bambulab/BambuStudio#4634

franciscoglop, 2024-08-15:
Same here, as said above it should be some package update in Arch. Still working Bambu Studio for me though

hyperpuncher, 2024-08-16:
Downgrading **freetype2** to `2.13.2-2` fixed this for me.

felipebalbi, 2024-08-16:
same here, Arch Linux

Eris-JBG, 2024-08-16:
Can also confirm that downgrading freetype2 fixes this issue!

xdevs23, 2024-08-18:
How can such an update cause an entire application to segfault? I don't know if freetype follows semver, but if they do, this little patch update shouldn't break anything. The question is, who is responsible for the breakage in the first place? Is it freetype, arch or orca? I can confirm downgrading freetype2 fixes the issue.

eris-price, 2024-08-18:
Trying to assign blame like that isn’t helpful. Sometimes things break and it’s no one’s fault, esp since arch isn’t even officially supported. the freetype project dates back to 1996, so they may follow some type of semver but it’s not a perfect system, and you don’t always know when a change is breaking. freetype2 is from the main arch repos, and it’s gonna be tested against other stuff in the main repos, but it can’t and shouldn’t be tested against every aur package.

Eris-JBG, 2024-08-18:
Trying to assign blame like that isn’t helpful. Sometimes things break and it’s no one’s fault, esp since arch isn’t even officially supported. the freetype project dates back to 1996, so they may follow some type of semver but it’s not a perfect system, and you don’t always know when a change is breaking. freetype2 is from the main arch repos, and it’s gonna be tested against other stuff in the main repos, but it can’t and shouldn’t be tested against every aur package.

lemzwerg, 2024-08-18:
Looking at the gdb output I see that the crash is caused by a return value from Cairo and/or Pango while getting information on a font with a 'COL' v1 table (most likely an Emoji font).  You should thus first contact the Cairo people asking for help – of course it is also possible that this is a problem of OrcaSlicer (I have no idea).  To do that efficiently you should build the crashing program without any optimizations to get better stack traces – in particular, the topmost frame (#0) doesn't give any useful information, which means it is not clear at all *where* the program crashes.

aeonik, 2024-08-18:
> How can such an update cause an entire application to segfault? I don't know if freetype follows semver, but if they do, this little patch update shouldn't break anything. The question is, who is [...]

Running Arch and other bleeding distros is nice because it gives us the latest software, but it's called bleeding edge because we get cut sometimes. 

It's nice though because we find these issues and can get this information to the developers that build our software, so they know to fix it.

We are the canaries of the software coal mines. It's not glamours, it's frustrating, but somebody has to do it.

Regards,
Aeonik

xdevs23, 2024-08-18:
> Trying to assign blame like that isn’t helpful. Sometimes things break and it’s no one’s fault, esp since arch isn’t even officially supported. the freetype project dates back to 1996, so they may [...]

I understand and I'm not trying to put a blame on anyone. For lack of better words my comment turned out to appear negative. I apologize for that. I'm just trying to understand which component is causing the segfault – is it freetype2 with a bug or breaking change or is it Orca for using something in freetype2 in a way that didn't cause issues before but now does, or is it an Arch specific issue. It's really just a question.

xdevs23, 2024-08-18:
> Running Arch and other bleeding distros is nice because it gives us the latest software, but it's called bleeding edge because we get cut sometimes.
>
> It's nice though because we find these issues and can get this information to the developers that build our software, so they know to fix it.
> 
> We are the canaries of the software coal mines. It's not glamourous, it's frustrating, but somebody has to do it.

I agree with that. It gives us the opportunity to fix things before they arise at users more downstream or with non-rolling distros.

apodtele, 2024-08-19:
`_cairo_ft_scaled_glyph_is_colr_v1` straightforwardly calls `FT_Get_Color_Glyph_Paint`. The trace is lost there because FreeType's debug information is missing. In other words the backtrace needs to be improved once the missing package is installed or FreeType is recompiled with `-g`. 

Don't despair, improve the backtrace.

Sanaki, 2024-08-19:
While imperfect, this should be a slightly improved backtrace.
```
Thread 1 "orcaslicer_main" received signal SIGSEGV, Segmentation fault.
0x000055555913de74 in tt_face_get_paint_layers ()
(gdb)  thread apply all bt

Thread 7 (Thread 0x7fffdda006c0 (LWP 2945021) "pool-orca-slice"):
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007ffff6dc7367 in g_cond_wait_until (cond=<optimized out>, mutex=0x55555bdfef00, end_time=<optimized out>) at ../glib/glib/gthread-posix.c:1677
#2  0x00007ffff6d348d5 in g_async_queue_pop_intern_unlocked (queue=0x55555bdfef00, wait=1, end_time=2075481119181) at ../glib/glib/gasyncqueue.c:378
#3  0x00007ffff6da127b in g_thread_pool_wait_for_new_task (pool=0x55555bdf1170) at ../glib/glib/gthreadpool.c:260
#4  g_thread_pool_thread_proxy (data=<optimized out>) at ../glib/glib/gthreadpool.c:325
#5  0x00007ffff6d9b266 in g_thread_proxy (data=0x7fffd8015bb0) at ../glib/glib/gthread.c:835
#6  0x00007fffef8a339d in start_thread (arg=<optimized out>) at pthread_create.c:447
#7  0x00007fffef92849c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

Thread 6 (Thread 0x7fffde4006c0 (LWP 2945020) "gdbus"):
#0  0x00007fffef91a63d in __GI___poll (fds=0x7fffc4000b90, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff6dcd92d in g_main_context_poll_unlocked (priority=2147483647, context=0x55555be0c020, timeout=<optimized out>, fds=0x7fffc4000b90, n_fds=2) at ../glib/glib/gmain.c:4521
#2  g_main_context_iterate_unlocked.isra.0 (context=0x55555be0c020, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4212
#3  0x00007ffff6d6c7b7 in g_main_loop_run (loop=0x55555ba1fa10) at ../glib/glib/gmain.c:4419
#4  0x00007ffff6fce574 in gdbus_shared_thread_func (user_data=0x55555be0bff0) at ../glib/gio/gdbusprivate.c:284
#5  0x00007ffff6d9b266 in g_thread_proxy (data=0x55555be0c120) at ../glib/glib/gthread.c:835
#6  0x00007fffef8a339d in start_thread (arg=<optimized out>) at pthread_create.c:447
#7  0x00007fffef92849c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

Thread 5 (Thread 0x7fffdee006c0 (LWP 2945019) "gmain"):
#0  0x00007fffef91a63d in __GI___poll (fds=0x55555b48fea0, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff6dcd92d in g_main_context_poll_unlocked (priority=2147483647, context=0x55555be007c0, timeout=<optimized out>, fds=0x55555b48fea0, n_fds=2) at ../glib/glib/gmain.c:4521
#2  g_main_context_iterate_unlocked.isra.0 (context=context@entry=0x55555be007c0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4212
#3  0x00007ffff6d6afc5 in g_main_context_iteration (context=0x55555be007c0, may_block=may_block@entry=1) at ../glib/glib/gmain.c:4282
#4  0x00007ffff6d6b01a in glib_worker_main (data=<optimized out>) at ../glib/glib/gmain.c:6442
#5  0x00007ffff6d9b266 in g_thread_proxy (data=0x55555be009c0) at ../glib/glib/gthread.c:835
#6  0x00007fffef8a339d in start_thread (arg=<optimized out>) at pthread_create.c:447
#7  0x00007fffef92849c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

Thread 4 (Thread 0x7fffe02006c0 (LWP 2945018) "pool-spawner"):
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007ffff6dc6a10 in g_cond_wait (cond=0x55555bdfed58, mutex=0x55555bdfed50) at ../glib/glib/gthread-posix.c:1552
#2  0x00007ffff6d3490c in g_async_queue_pop_intern_unlocked (queue=0x55555bdfed50, wait=1, end_time=-1) at ../glib/glib/gasyncqueue.c:375
#3  0x00007ffff6da06b7 in g_thread_pool_spawn_thread (data=<optimized out>) at ../glib/glib/gthreadpool.c:297
#4  0x00007ffff6d9b266 in g_thread_proxy (data=0x55555bdfdf90) at ../glib/glib/gthread.c:835
#5  0x00007fffef8a339d in start_thread (arg=<optimized out>) at pthread_create.c:447
#6  0x00007fffef92849c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

Thread 3 (Thread 0x7fffdf8006c0 (LWP 2945012) "orcaslicer_main"):
#0  FcCompareValueList (object=50, match=0x7ffff69b6610 <_FcMatchers+1200>, v1orig=0x55555c168390, v2orig=<optimized out>, bestValue=bestValue@entry=0x0, value=value@entry=0x7fffdc6392f8, n=0x0, result=0x7fffdf7f2a20) at ../fontconfig/src/fcmatch.c:435
#1  0x00007ffff69886b8 in FcCompare (pat=pat@entry=0x55555bfa2ba0, fnt=0x7fffe0534620, value=value@entry=0x7fffdc6392f8, result=result@entry=0x7fffdf7f2a20, data=data@entry=0x7fffdf7f2978) at ../fontconfig/src/fcmatch.c:647
#2  0x00007ffff698ed94 in IA__FcFontSetSort (config=<optimized out>, sets=sets@entry=0x55555bfa2c10, nsets=nsets@entry=1, p=0x55555bfa2ba0, trim=trim@entry=1, csp=csp@entry=0x0, result=0x7fffdf7f2a20) at ../fontconfig/src/fcmatch.c:1275
#3  0x00007ffff695874b in sort_in_thread (task_data=0x55555bfa2c00) at ../pango/pango/pangofc-fontmap.c:932
#4  fc_thread_func (data=0x55555b3b0d00) at ../pango/pango/pangofc-fontmap.c:972
#5  0x00007ffff6d9b266 in g_thread_proxy (data=0x55555b3b03b0) at ../glib/glib/gthread.c:835
#6  0x00007fffef8a339d in start_thread (arg=<optimized out>) at pthread_create.c:447
#7  0x00007fffef92849c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

Thread 1 (Thread 0x7fffeb24eb80 (LWP 2944172) "orcaslicer_main"):
#0  0x000055555913de74 in tt_face_get_paint_layers ()
#1  0x00007ffff716778d in _cairo_ft_scaled_glyph_is_colr_v1 (scaled_font=<optimized out>, scaled_glyph=0x55555c1e8730, face=0x55555bfa32e0) at ../cairo/src/cairo-ft-font.c:3266
[... cut ...]
#24 0x00005555596bf9d7 in wxGCDCImpl::DoGetTextExtent(wxString const&, int*, int*, int*, int*, wxFont const*) const ()
#25 0x0000555556b7429a in Slic3r::GUI::SplashScreen::Decorate(wxBitmap&) ()
#26 0x0000555556b69dc5 in Slic3r::GUI::GUI_App::on_init_inner() ()
#27 0x0000555556b6b62a in Slic3r::GUI::GUI_App::OnInit() ()
#28 0x00005555599cd4d7 in wxEntry(int&, wchar_t**) ()
#29 0x0000555556b2a01b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#30 0x0000555555fc442b in Slic3r::CLI::run(int, char**) ()
#31 0x0000555555ec3209 in main ()
```

lemzwerg, 2024-08-19:
Thanks, but the most crucial information is still missing, namely FreeType's file and line number details for frame #0.

If possible, please compile FreeType manually without optimization and temporarily replace the OS's version.

Sanaki, 2024-08-19:
I tried my hand at that, but I must have been doing something wrong, because I couldn't get it to read the symbols, even after installing the -debug package. Hopefully someone who's done this before can give it a shot.

ahmet-cetinkaya, 2024-08-20:
> Downgrading **freetype2** to `2.13.2-2` fixed this for me.

[image]

eteq, 2024-08-24:
I'm seeing this problem in the appimage same as the OP and others... but I do *not* get the segfault when I compiled it using an Arch Linux PKGBUILD - specifically, I took https://aur.archlinux.org/packages/orca-slicer, updated the tagged version to 2.1.1, commented out some of the patches that were no longer relevant, and than a `makepkg -si` yielded a working Orca Slicer!

(it could still be a threading issue of course - by compiling it that way I might have just resulted in some changed linking order that works around whatever race condition is the problem.  But at least it got me a working orca slicer on Arch...)

deddroid, 2024-08-24:
same issue for me. downgrading freetype2 to 2.13.2-2 worked to open the app, and backup my configs, but I have no view in 3D area

wuhgit, 2024-08-29:
Same issue, after some testing, I can start OrcaSlicer successfully with the following command on ArchLinux

```
WEBKIT_DISABLE_DMABUF_RENDERER=1 __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json /opt/orca-slicer/bin/orca-slicer
```

icoz, 2024-09-02:
> same issue for me. downgrading freetype2 to 2.13.2-2 worked to open the app, and backup my configs, but I have no view in 3D area.
> 
> also the backup is impossible cos I can't expand the windows to select my settings to export

Same issue for me. 
But I've just recompile freetype2 package.
`asp checkout freetype2`
`cd freetype2/trunk`
`makepkg -sci`

Package orca-slicer-bin-2.1.1-1 work fine with freetype2-2.13.0-1.

jrobbom5, 2024-09-05:
> Package orca-slicer-bin-2.1.1-1 work fine with freetype2-2.13.3-1.

This combo with the recompiled freetype2 module no longer appears to work, it appears to do the same thing as freetype2-2.13.2-2 was doing: ie Orca starts fine, and you can see the menus on the left, but the build late area on the right is completely blank.

Communing-With-Coyotes, 2024-09-28:
Tried the environment variables listed by @wuhgit and downgraded freetype to 2.13.2-2 and while Orca will launch, it freezes shortly after (usually within 10-15 seconds) and crashes roughly another 10-15 seconds after that. Lots of Gtk-CRITICAL errors, but those aren't unusual. Even when it worked properly, Orca has always spit hundreds of these errors.

xdevs23, 2024-09-30:
@Communing-With-Coyotes 
> Tried the environment variables listed by @wuhgit and downgraded freetype to 2.13.2-2 and while Orca will launch, it freezes shortly after (usually within 10-15 seconds) and crashes roughly another [...]

The crashes can be worked around by editing the connection settings and changing the Device UI to something it can't reach:
[image]

It was crashing for me otherwise with the same symptoms as you described.

Sanaki, 2024-10-01:
Looks like the latest beta has fixed this issue, perhaps simply through a build system update. Used `OrcaSlicer_Linux_Ubuntu2404_V2.2.0-beta26.AppImage` and it started up a-ok with no modifications.

jordicunillerarivera, 2024-10-03:
For me installing Webkit2gtk, gst-plugins-base, and gst-libav from AUR fix the crash. The last one is only needed
for the X1C camera.

anarsoul, 2024-11-22:
Can you try flatpak from https://github.com/SoftFever/OrcaSlicer/issues/3949#issuecomment-2483844840 ?
I'm running it on Arch and it works fine

agsimmons, 2024-11-22:
This issue was resolved for me across two different computers (Arch and EndevourOS) with orcaslicer 2.2.0 and freetype2 2.13.3-1.

Previously, I experienced the same crash on both computers With orcaslicer 2.1.1, and downgrading to freetype2 2.13.2-2 resolved the issue on both.

molostovvs, 2025-01-09:
```
yay -Rns ttf-harmonyos-sans
```

fixes crash for me

closed 2025-04-17 by github-actions[bot]