OrcaSlicer/OrcaSlicer #6473: Picasso, repetier

Mirniy2000, 2024-08-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Я расстраиваюсь когда не  вижу что разработчики добавили профиль принтеров Picasso 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Подружить свой принтер Picasso со слайсером 

### Describe alternatives you've considered

1. Печать на всех предоставленных разработчиком прошивках!

NiXTheDev, 2024-08-16:
English translation:

<...>

# Is your feature request related to a problem?
I'm upset when I don't see that the developers have added a profile for Picasso* printers

<...>

# Describe the solution you'd like
Connect my Picasso* printer with a slicer

<...>

* Picasso3D - a Russian based company making 3D printers, all of their models have naming scheme of: Picasso Designer <x>, where <x> is the variant of the printer, namely: "Pro", "Pro 2", "XL", "XL 2" or nothing, these printers use a proprietary firmware, but most of the Repetier/Marlin GCode spec is supported(basic commands like G0/1 or T0-9, M190, 140, 104, 109 are supported, Arcs of G2/3 are unknown, but should be supported)

NiXTheDev, 2024-08-16:
Examples of the PolygonX generated gcode for all available Picasso3D Designer printers:
[3dbenchy-DesignerXL-Default-0.4mmnozzle.txt](https://github.com/user-attachments/files/16642244/3dbenchy-DesignerXL-Default-0.4mmnozzle.txt)
[3dbenchy-DesignerXLPro-Default-Double0.4mmnozzle.txt](https://github.com/user-attachments/files/16642246/3dbenchy-DesignerXLPro-Default-Double0.4mmnozzle.txt)
[3dbenchy-DesignerXLProSeries2-Default-0.4mmnozzle.txt](https://github.com/user-attachments/files/16642247/3dbenchy-DesignerXLProSeries2-Default-0.4mmnozzle.txt)
[3dbenchy-DesignerXLProSeries2-Default-Double0.4mmnozzle.txt](https://github.com/user-attachments/files/16642249/3dbenchy-DesignerXLProSeries2-Default-Double0.4mmnozzle.txt)
[3dbenchy-DesignerXPro-Default-Double0.4mmnozzle.txt](https://github.com/user-attachments/files/16642250/3dbenchy-DesignerXPro-Default-Double0.4mmnozzle.txt)
[3dbenchy-DesignerXProSeries2-Default-Double0.4mmnozzle.txt](https://github.com/user-attachments/files/16642251/3dbenchy-DesignerXProSeries2-Default-Double0.4mmnozzle.txt)
[3dbenchy-DesignerXSeries2-Default-0.4mmnozzle.txt](https://github.com/user-attachments/files/16642252/3dbenchy-DesignerXSeries2-Default-0.4mmnozzle.txt)
[3dbenchy-DesignerClassic-Default-0.4mmnozzle.txt](https://github.com/user-attachments/files/16642253/3dbenchy-DesignerClassic-Default-0.4mmnozzle.txt)
[3dbenchy-DesignerX-Default-0.4mmnozzle.txt](https://github.com/user-attachments/files/16642254/3dbenchy-DesignerX-Default-0.4mmnozzle.txt)

Notes:
* Pro variants of Designer printers have 2 nozzles that they can switch between
* All Picasso Designer printers support nozzles of sizes: 0.3mm, 0.4mm, 0.5mm, and 0.8mm
* All Picasso Designer printers can be monitored and controlled over LAN, the printers have an ethernet port and no wifi capability, as well as a USB-B(like on paper printers) port either on the front or the back of the printer depending on the model, so far it is unknown why the USB port is included on the printer, but it is assumed to just be for printing over serial like on many printers that run Marlin or other firmware that supports printing over serial connection

Mirniy2000, 2024-08-17:
Претензий к данному принтеру  нет от слова совсем но если только круговую интерполяцию не поддерживает G2,G3 в данный момент считаю что сильно отстает PolygonX в том плане что отсутствуют возможность добавить скрипты для теста пластика, поэтому и обратился за помощью orca слайсера так как знаю что orca дитя  prusa слайсера и быстрее развивается чем CURA!

Mirniy2000, 2024-08-18:
Что это за g-коды , для чего они?

NiXTheDev, 2024-08-19:
> Что это за g-коды , для чего они?

Those are examples for SoftFever and contributors to perhaps implement a set of commands for those printers in Orca

Mirniy2000, 2024-08-26:
Приветствую! Все отлично распечаталось спасибо, как я понимаю для создания профиля нужно определенное количество запросов на эту тему?

cochcoder, 2024-10-02:
Hi, I can't get the slicer running on my system. Can you send me the configurations for the printers and their filaments? Also, which one of these printers can you personally test?

Mirniy2000, 2024-10-04:
Привет! Что именно должно входить в перечень конфигураций? У меня в наличие принтер Picasso designer x pro series 2.

cochcoder, 2024-10-04:
> Привет! Что именно должно входить в перечень конфигураций? У меня в наличие принтер Picasso designer x pro series 2. 

If you can include all the print settings, filament settings, and machine settings for all of the printers, that would be great! I need to finish up a few things first, but once I'm done I'll be working on this. 

I'll start with the x pro series 2, and if you're willing, you can test the profiles I add. Just to make sure that everything ports over nicely to Orca Slicer.

Mirniy2000, 2024-10-04:
Правильно я понимаю что мне нужно сформировать задание в PoligonX с максимальным количеством настроек и прислать Вам? Или лучше прислать последнюю  версию прошивки принтера и профили материала? 

Просто большинство настроек такие как охлаждение или температура  сопла и стола находятся в самом принтере!

cochcoder, 2024-10-04:
You'll need to send me all the configurations from PoligonX itself. The printer's firmware from the website is not in a human readable format.

Mirniy2000, 2024-10-07:
Не получается скинуть здесь файл

cochcoder, 2024-10-07:
Is the file size too big? If so, you can create a project on GitHub, upload the files there, and share the link to it.

Mirniy2000, 2024-10-08:
Не поддерживает формат G код (PolygonX Task File .plgx), может лучше на почту скинуть?

cochcoder, 2024-10-10:
Sure, here's a email you can send it to: github.filter194@slmail.me

AlexSidorov, 2024-11-26:
I got a Picasso Xpro of the first series at work. And I would also like to have the ability to cut for printing in this printer in ORCA Slicer. Here they ask about configuration files, etc., so I can offer a link to download and install the Designer program. It is freely distributed and you can immediately create/save plgx files there. Accordingly, you can play with the settings of their program and see what settings and in what form are exported to plgx.

I want to say right away that their program can directly translate regular GCODE into their plgx format, but in this case it is impossible to change anything in the settings (add support, for example)..

Therefore, if there is some kind of profile in ORCA for this printer, it would be great.

Thanks

Mirniy2000, 2024-11-27:
Я думаю в ближайшее время не получить профиль данных принтеров в Orca, так как производитель отказывается открывать код ПО.

AlexSidorov, 2024-11-27:
> Я думаю в ближайшее время не получить профиль данных принтеров в Orca, так как производитель отказывается открывать код ПО.

Тут можно пойти другим путем. Правда он немного муторный, но все же. Создаем тело - прямоугольник. Далее режем его в Orca, открываем в Designer и этот конвертер из готового GCODE переводит в свой формат. Таким образом можно будет сопоставить исходник STL, настройки ORCA , готовый GCODE и PLGX. Основное (для меня) это мосты, которые увы и ах в Designer не предусмотрены. Я в принципе готов помочь чем могу. Только вот чем?

Mirniy2000, 2024-11-28:
Я добился такого результата что принтер все делает правильно только без подачи материала.

AlexSidorov, 2024-11-29:
> Я добился такого результата что принтер все делает правильно только без подачи материала.

А не подскажете как? Вот у них написано, что ....

Prusa Slicer

Создайте собственный профиль принтера с помощью Мастера настройки (Configuration Assistant / Configuration Wizard), выбрав в нем Пользовательский принтер (Custom Printer). Задайте имя пользовательского профиля, выберите тип прошивки, установите размеры рабочей зоны и диаметры сопел в соответствии с выбранным устройством. Нажмите “Завершить” (“Finish”).

При подготовке G-code для Designer Classic учитывайте наличие фиксаторов стекла. Располагайте модель на платформе с отступом — не менее 10 мм спереди.

После настройки профиля принтера убедитесь, что вы используете эти пользовательские G-коды в меню Настройки принтера > Продвинутый (Printer Settings > Expert mode):

Тип G-кода (G-code flavour):
Repetier

Стартовый G-код (Start G-code):
G28

G-код, выполняемый перед сменой слоя (Before layer change G-code):
M532 X0 L[layer_num] Z[layer_z]

Эта команда позволяет использовать систему контроля подачи пластика. Мы рекомендуем провести ее калибровку перед началом использования нового материала.

G-код, выполняемый при смене инструмента (Tool change G-code):
T1{next_extruder} I5 S1 R1 P1
﻿
где I — длина прочистки, S — скорость прочистки, R — длина ретракта, P — скорость ретракта. После переключения сопел активное сопло прочищается в специальном очистителе. Таким образом, печать дополнительных башен для очистки не требуется. Кроме того, возможно использование быстрого переключения сопел без прочистки. Чтобы активировать опцию, включите Быстрое переключение в меню принтера Настройки > Основные настройки.

Другие поля для ввода G-кодов заполнять не нужно, оставляйте их пустыми.

Cura Slicer

Создайте собственный профиль принтера, выбрав в строке меню Preferences > Configure Cura > Printers. Нажмите кнопку “Add” и выберите из списка Custom FFF printer. Введите имя принтера в поле Printer name и нажмите “Add”.

В появившемся окне Machine Settings настройте профиль следующим образом:

Printer Settings:
X (Width) — 200 mm (для Designer X / Designer X PRO / Designer Classic); 360 mm (для Designer XL / Designer XL PRO);
Y (Depth) — 200 mm (для Designer X / Designer X PRO / Designer Classic); 360 mm (для Designer XL / Designer XL PRO);
Z (Hight) — 210 mm (для Designer X / Designer X PRO / Designer Classic); 610 mm (для Designer XL / Designer XL PRO).

При подготовке G-code для Designer Classic учитывайте наличие фиксаторов стекла. Располагайте модель на платформе с отступом — не менее 10 мм спереди.

G-code flavour:
Repetier

Printhead Settings (для Designer X PRO / Designer XL PRO):

Number of Extruders: 2

Start G-code:
G28

Во вкладке Extruder 1 / Extruder 2 (для Designer X PRO / Designer XL PRO) установите диаметр сопла (Nozzle size) в соответствии с установленным на принтере и диаметр пластиковой нити (Compatible material diameter) — 1.75 mm.

Extruder 1 Start G-code:
T10 I5 S1 R1 P1

Extruder 2 Start G-code (для Designer X PRO / Designer XL PRO):
T11 I5 S1 R1 P1

где I — длина прочистки, S — скорость прочистки, R — длина ретракта, P — скорость ретракта. После переключения сопел активное сопло прочищается в специальном очистителе. Таким образом, печать печать дополнительных башен для очистки не требуется. Кроме того, возможно использование быстрого переключения сопел без прочистки. Чтобы активировать опцию, включите Быстрое переключение в меню принтера Настройки > Основные настройки.

Добавьте специальный скрипт, выбрав в строке меню Extensions > Post processing > Modify G-code. Нажмите “Add a script” и выберите Search and Replace. Впишите следующие скрипты:

Search:
;LAYER:

Replace:
M532 X0 L
﻿
Эти команды позволяют использовать систему контроля подачи пластика. Мы рекомендуем провести ее калибровку перед началом использования нового материала.

Другие поля для ввода G-кодов заполнять не нужно, оставляйте их пустыми.

Mirniy2000, 2024-11-29:
Вам это в Orca нужно или Prusa?

AlexSidorov, 2024-12-05:
Уважаемый @Mirniy2000  Вам не будет сложно связаться со мной по почте alexsidorov@mail.ru? Я был вчеар в офице Picasso, поговорил с начальником технической поддержки и судя по всему поддержки софта от них врядли дождешься, а вот печатать нужно. Спасибо

cochcoder, 2025-03-11:
Still looking into this. 

Sorry about the long pause, I plan on coming back to this soon.

Mirniy2000, 2025-03-30:
Было бы  интересно и дальше  с Вами  заниматься на эту тему!