OrcaSlicer/OrcaSlicer #6478: Changing the flow rate changes the flow in the preview quadratically

danik-ik, 2024-08-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.3 Cinnamon

### Additional system information

System:
  Kernel: 6.8.0-40-generic x86_64 bits: 64 compiler: N/A Desktop: Cinnamon 6.0.4 tk: GTK 3.24.33
    wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop Mobo: ASRock model: A520M-ITX/ac serial: <superuser required>
    UEFI: American Megatrends v: P1.10 date: 08/24/2020
CPU:
  Info: 8-core model: AMD Ryzen 7 PRO 4750G with Radeon Graphics bits: 64 type: MT MCP
    smt: enabled arch: Zen 2 rev: 1 cache: L1: 512 KiB L2: 4 MiB L3: 8 MiB
  Speed (MHz): avg: 1881 high: 4289 min/max: 400/4454 cores: 1: 400 2: 400 3: 4286 4: 2864
    5: 400 6: 400 7: 4117 8: 3459 9: 400 10: 4289 11: 400 12: 400 13: 400 14: 400 15: 4178 16: 3305
    bogomips: 114989
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: AMD Renoir driver: amdgpu v: kernel pcie: speed: 8 GT/s lanes: 16 ports:
    active: DP-1,HDMI-A-1 empty: DP-2 bus-ID: 07:00.0 chip-ID: 1002:1636 class-ID: 0300
  Device-2: Logitech Webcam C270 type: USB driver: snd-usb-audio,uvcvideo bus-ID: 3-2:2
    chip-ID: 046d:0825 class-ID: 0102 serial: <filter>
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: amdgpu,ati
    unloaded: fbdev,modesetting,vesa gpu: amdgpu display-ID: :0 screens: 1
  Screen-1: 0 s-res: 3840x1080 s-dpi: 96 s-size: 1016x285mm (40.0x11.2") s-diag: 1055mm (41.5")
  Monitor-1: DisplayPort-0 mapped: DP-1 pos: right model: Philips PHL 243V7 serial: <filter>
    res: 1920x1080 hz: 60 dpi: 93 size: 527x296mm (20.7x11.7") diag: 604mm (23.8") modes:
    max: 1920x1080 min: 720x400
  Monitor-2: HDMI-A-0 mapped: HDMI-A-1 pos: primary,left model: Philips PHL 243V7
    serial: <filter> res: 1920x1080 hz: 60 dpi: 93 size: 527x296mm (20.7x11.7") diag: 604mm (23.8")
    modes: max: 1920x1080 min: 720x400
  OpenGL: renderer: RENOIR (renoir LLVM 15.0.7 DRM 3.57 6.8.0-40-generic)
    v: 4.6 Mesa 23.2.1-1ubuntu3.1~22.04.2 direct render: Yes
Audio:
  Device-1: AMD Renoir Radeon High Definition Audio driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 07:00.1 chip-ID: 1002:1637 class-ID: 0403
  Device-2: AMD Family 17h HD Audio vendor: ASRock driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 07:00.6 chip-ID: 1022:15e3 class-ID: 0403
  Device-3: Logitech Webcam C270 type: USB driver: snd-usb-audio,uvcvideo bus-ID: 3-2:2
    chip-ID: 046d:0825 class-ID: 0102 serial: <filter>
  Sound Server-1: ALSA v: k6.8.0-40-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: ASRock driver: r8169
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 03:00.0 chip-ID: 10ec:8168
    class-ID: 0200
  IF: enp3s0 state: up speed: 100 Mbps duplex: full mac: <filter>
  Device-2: Intel Dual Band Wireless-AC 3168NGW [Stone Peak] driver: iwlwifi v: kernel pcie:
    speed: 2.5 GT/s lanes: 1 bus-ID: 04:00.0 chip-ID: 8086:24fb class-ID: 0280
  IF: wlp4s0 state: up mac: <filter>
  IF-ID-1: br-440a152ade08 state: down mac: <filter>
  IF-ID-2: br-68923040f141 state: down mac: <filter>
  IF-ID-3: br-6b8a6516ba64 state: up speed: 10000 Mbps duplex: unknown mac: <filter>
  IF-ID-4: br-b6e2661fdbba state: down mac: <filter>
  IF-ID-5: br-e4275738629e state: up speed: 10000 Mbps duplex: unknown mac: <filter>
[... 89 lines cut ...]
Switch preview color scheme to "Flow". 
Remember the value
Set filament→flow rate to 0.5. Slice the model.
Switch preview color scheme to "Flow". 
Compare new value with previous.

### Actual results

The values ​​differ by a factor of four

### Expected results

The values ​​differ by a factor of two

### Project file & Debug log uploads

[image]
[image]
Note that REAL flows differ by a factor of two. The issue is about preview, not g-code
[flow.zip](https://github.com/user-attachments/files/16638789/flow.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

henrivdr, 2024-08-17:
Thanks for raising this one.  I saw the same, but I am glad to see the real ratio is correct.

Rickthebig, 2024-08-19:
I noticed a similar,  but opposite issue with the settings "Top surface flow ratio" and " Bottom surface flow ratio" - both need to be changed to a larger extend  than expected, in order to achieve a minor change of the corresponding flow rate. I cannot quantify the deviation, but something is definitely wrong here too.

igiannakas, 2024-08-20:
It is caused by this commit here: 
https://github.com/SoftFever/OrcaSlicer/pull/5628

More details in the comment in the commit above.

closed 2024-08-22 by SoftFever