OrcaSlicer/OrcaSlicer #6480: WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect.

WimVDK, 2024-08-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

11 23H2

### Printer

X1C

### How to reproduce

1. Load project file
2. Set lids as 1 color and boxes as another
3. slice with print sequence "by layer"

### Actual results

Slicing fails with error WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect.

[image]

### Expected results

Proper slicing. Slicing only works when set to "by object"

### Project file & Debug log uploads

[boxes with lids.zip](https://github.com/user-attachments/files/16645209/boxes.with.lids.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ash286, 2024-11-07:
Has this bug fixed in latest v2.2.0 official? I'm keep encountering with this error of "Error: WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect."

hanakin, 2024-11-30:
i on mac with version 2.2.0 and am getting this error as well when adding text to an object

rozhkovets, 2025-01-24:
I get the same error if the model consists of several parts and was split into models. If the model is re-merged - there is no error.

W1SPS, 2025-02-25:
I get this error when I only have 2 filaments selected. One as the main color and the other as a top surface color fill for .005 indented text. And this is on my X1C that until last thursday didn't do this.

SvenCarstensen, 2025-04-24:
The error still occurs in version 2.3.0 on a Prusa XL.

closed 2025-08-01 by github-actions[bot]