OrcaSlicer/OrcaSlicer #6484: Painting overrides text modifiers and parts

ktiedt, 2024-08-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

14.5

### Printer

Bambu Lab X1C

### How to reproduce

1. Add primitive -> cube
2. Right click cube, add modifier -> text
3. Set filament color for modifier to something other than the cubes color.
4. Slice project, cube will have flush text in the color selected.
5. Click back to Prepare
6. Select cube
7. Click paint
8. Paint face of cube with text modifier 3rd color.
9. Slice project again
10. You can change modifier type from 'modifier' to 'part' and the same problem occurs

It seems impossible to have a model region designed for text, with an easily paintable background as well.

### Actual results

Painted color overwrites all attempts at displaying flush text on the painted surface.

### Expected results

Text is printed flush with surrounding background region printed in painted color.

### Project file & Debug log uploads

[Paint + Flush text bug.3mf.zip](https://github.com/user-attachments/files/16646715/Paint.%2B.Flush.text.bug.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]
[image]

AB cube is no paint, XY cube is paint, you can switch the text modifer to a part, and it will also fail in the same manner.

closed 2024-11-24 by github-actions[bot]

NeonWizard, 2025-05-05:
Bump on this please

noiob, 2025-05-14:
just ran into this, imo this is a bug and should be reopened. If anything the software should warn me