OrcaSlicer/OrcaSlicer #6486: v2.2 Nightly Dev's crashing when editing (custom) Printer Settings

Ergonomicmike, 2024-08-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-dev

### Operating System (OS)

Windows

### OS Version

Win10

### Printer

Ender 3 v2

### How to reproduce

Open a recent (August 13 or 14) 2.2.0-dev portable Nightly.
Change some text in Printer settings>Machine G-code. Or change or add to a Note.
Click away to another tab in Printer settings.

### Actual results

Crash in WIn10.
Sometimes Windows displays a message about a memory problem.

### Expected results

In the past I have edited my Printer settings without issue. This problem just started.

### Project file & Debug log uploads

[Orca crash log.zip](https://github.com/user-attachments/files/16646775/Orca.crash.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Okay, when I see Windows report a memory problem, it could be my RAM. I will run a Diagnostic after posting here.
If it's not a RAM problem on my end, it looks like this started with v2.2.0. (I just now tried an August 8 Nightly. Same behavior.)
I just now tried v2.2.1-portable. It does not crash when editing the same Printer Setting.

VoltaicShock, 2024-08-18:
I don't think it's your RAM. I am having this issue on Linux as well with the .appImage.

I thought it might just be my machine or Linux but it seems you are having the issue too. It closed on me before I could save my project, fortunately it recognized that it did this and let me pull it back up when I restarted OrcaSlicer.

Ergonomicmike, 2024-08-19:
@VoltaicShock Thank you for confirming that it's not just me.

For the record, I performed a memtest and my RAM tests okay.

StarkRG, 2024-08-19:
I'm getting this on the 2.1.1 Linux release, not Nightly. This is on Pop!_OS 22.04

Weirdly, there's nothing written to the debug log as it crashes, what's written is already there. 

[debug_Mon_Aug_19_21_05_23_699455.log](https://github.com/user-attachments/files/16659438/debug_Mon_Aug_19_21_05_23_699455.log)

Nor does it happen *every* time you open the Printer Settings window, but you can pretty reliably get it to crash by opening the settings window and then swapping between tabs. It seems to take more tab switching in non-advanced mode, but does eventually happen. The filament settings window does not seem to have a similar issue (switching tabs a lot didn't cause a crash).

Noisyfox, 2024-08-21:
This should have been fixed in #6261, please try the latest nightly build.

Ergonomicmike, 2024-08-21:
Still crashing for me. I downloaded the latest portable as of this writing (two days ago). 

I open Printer, change some text in the G-code tab, move to another tab > crash.

See video.

[Orca crash.zip](https://github.com/user-attachments/files/16695775/Orca.crash.zip)

Noisyfox, 2024-08-22:
You are right, I could also reproduce this myself. Will take a look asap.

Noisyfox, 2024-08-22:
https://github.com/SoftFever/OrcaSlicer/blob/6dc8b285294a3dd0fe4f3f560f70d2f0efc24a49/src/slic3r/GUI/Plater.cpp#L1389

This line cases this issue. However I don't have a proper solution at the moment.

Ergonomicmike, 2024-08-22:
I'm not a coder. But from the 50,000 ft view, perhaps a work around is to set a Tool Changer variable to a default of 1 for  all printers, regardless of whether they're capable of more then one tool changer.

PR #6537 (2024-08-23): Fix crash in printer config when switching tabs
This fixes #6486
closed 2024-08-23 by SoftFever