OrcaSlicer/OrcaSlicer #6489: Top layer only partially generated

mikljohansson, 2024-08-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Ubuntu

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

The attached ZIP file contains the STL and machine, filament and process configurations. Just import the model, config and slice it to see the issue on the top layer.

### Actual results

Importing the STL in OrcaSlicer 2.1.1 and slicing it with the configuration produces a partial top layer. See the screenshots and the gcode file contained in the ZIP

### Expected results

The top and bottom surfaces of the model are parallel, so I'd expect the top layer to be full

### Project file & Debug log uploads

[orca-partial-layer-bug-report.zip](https://github.com/user-attachments/files/16647869/orca-partial-layer-bug-report.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]
[image]
[image]
[image]

DVSVIDEO, 2024-08-18:
I downloaded your model, sliced and diced it and experienced no problems.
This could be your problem.
Setup or placement problem. It happens when the surface is slanted.
[image]

mikljohansson, 2024-08-18:
Thanks for the feedback! I notice that you're using a 0.2mm layer height and not the config I attached. I can only reproduce this issue when setting the layer height to 0.12mm or 0.16mm, perhaps you could try adjusting your layer height?

The top and bottom surfaces of the model are 100% parallel in the CAD program (I've even measured it there now), but I guess perhaps the STL export could be off (I'm using OnShape).

mikljohansson, 2024-08-18:
Indeed, it seems that the OnShape export was the issue. When exporting with Y-axis up the issue disappears, so it was some vertices that were bad in the STL. Thanks for helping me figure this out

closed 2024-08-18 by mikljohansson