OrcaSlicer/OrcaSlicer #6498: Outer perimeter overhangs should not be treated as bridges

kozross, 2024-08-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

[This video segment](https://youtu.be/HcSOz-Lsxgg?si=6o0AeqxVVAlMKqUD&t=100) shows the problem. When you have an overhang that's also an outer perimeter, OrcaSlicer treats it as a bridge, and forces it to be printed at bridge speed, which is often too fast, especially on steep overhangs.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Outer perimeter overhangs should be treated as overhangs.

### Describe alternatives you've considered

Manually slowing down the bridge speed to the speed of the slowest overhang fixes this, but it's inconvenient. You have to do it on a per-model basis, and the print time increases a lot when this happens.

TinyMito, 2024-08-19:
Test model you can use [here](https://makerworld.com/en/models/595369).

tome9111991, 2024-08-19:
enable "Slow down for curled perimeters" 
[image]

SoftFever, 2024-08-19:
I want to clarify: this isn't a bug.

When the overhang is more than a certain percentage (less than 10% supported area), it is considered a bridge by design. The difference arises from the default profile settings for MK4S in PS and P1/X1 in Orca. MK4(S) sets the bridge speed to 15mm/s, while the Bambu P1/X1 profile in Orca is set to 50mm/s.

The Bambu profiles in Orca pretty much use Bambu's profiles in BS as-is.

If you are using Orca's profiles for Voron or other machines, there's a greater chance you will have good results with this overhang test because in Orca, I set the default bridge speed to 25mm/s.
[image]

closed 2024-08-19 by SoftFever

igiannakas, 2024-08-20:
Updated tool tips explain this now in the latest night ;) https://github.com/SoftFever/OrcaSlicer/pull/6501

igiannakas, 2024-08-20:
The tooltip says it’s generally recommended to have the curled perimeters on unless your printer has adequate/strong cooling. It also has downsides, like artefacts from the significant slowdown if PA is not tuned properly. 

It’s all a trade off - there is no best setting…

It depends on what you’re printing and what the capabilities of the machine are, even down to what environment you’re printing in (room temperature for example). 

Personally I have this option on among other things but that is what works for what I typically print, in the environment I print plus with the material that I mostly use. 

The best way is really to experiment with your settings to match your specific individual needs :)

I do have my profiles published on discord but again, these are tailored to what I print…

matthew-humphrey, 2024-08-26:
> I want to clarify: this isn't a bug.
>
> When the overhang is more than a certain percentage (less than 10% supported area), it is considered a bridge by design. The difference arises from the default [...]

It seems like a bug, because a bridge by definition is a span between two supported points. That isn't the case here.

igiannakas, 2024-08-26:
A bridge is also a bridge a 100% overhang, so..

TinyMito, 2024-08-26:
FYI, Bambu Studio 1.9.4 added the following for overhang changes:

[image]

[image]

Gridzilla, 2024-09-04:
In case people reading this are looking where to change the default setting of "Bridge speed" and "Slow down for curled overhangs" for Bambu printers it's in this file: %APPDATA%\OrcaSlicer\system\BBL\process\fdm_process_bbl_common.json

Currently testing it set like this, I did not have the line "slowdown_for_curled_perimeters" at all and I had to add it. And I edited the default bridge speed. 
[image]

SoftFever, 2024-09-04:
> In case people reading this are looking where to change the default setting of "Bridge speed" and "Slow down for curled overhangs" for Bambu printers it's in this file: [...]

These options are available in the UI.
Did you use "classic mode"?  The "Slow down for curled perimeters" didn't work for classic mode
[image]

Gridzilla, 2024-09-04:
No I did not use classic mode. Yes these options are indeed in the UI, however when using AMS the default profiles are suggested for new filaments and it's easy to forget to change these two settings. 

By default the "Slow down for curled perimeters" is not set and bridge speed seems to be set at 50mm/s as per your screenshot. But that combination of settings gives bad overhang performance as was pointed out in the YT video that started this issue/thread. 

I am not saying this is the best, I just want people to know what options they have. I have a X1C, A1 and A1 Mini. Making edited copies of all the default profiles quickly becomes unmanageable.

SoftFever, 2024-09-04:
> It seems like a bug, because a bridge by definition is a span between two supported points. That isn't the case here.

The distinction between an overhang and a bridge is indeed somewhat fuzzy. 
This comes down to whether it's necessary to make a differentiation: if a 99% unsupported perimeter should be considered as an overhang or a bridge. In Orca and PrusaSlicer, it's treated as a bridge.

I'm not suggesting one approach is better or worse; I'm simply stating the current state of implementation, and it seems to be working fine so far. But nothing is set to stone.

Regarding the original issue, the differing results between the MK4S and Bambu printers are solely due to the bridge speed setting. 
If you set the bridge speed to 50 in PrusaSlicer, you'll get the same poor result. Conversely, if you set the bridge speed to 15 in Orca for any printer, you'll get the same good result.

Now, whether we should adjust the default bridge speed for Bambu printers to 15mm/s or 25mm/s, or keep it as is, is another question altogether – and one that's open to options.