OrcaSlicer/OrcaSlicer #6510: Retractions added to inside of Concentric Bottom Shell setting causing visible seam in bottom surface

skl111, 2024-08-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-DEV

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambu Labs P1 Series

### How to reproduce

Install this build of Orca slicer,
https://github.com/SoftFever/OrcaSlicer/actions/runs/10341055071
set concentric for the bottom surface on a cylindrical object
slice
and the retractions create a pattern that shows up as a gap in the bottom surface finished, looking like a scar.

Repeat same process with this build
https://github.com/SoftFever/OrcaSlicer/actions/runs/10320155058
and you get the expected result, no scar and smooth bottom finished with a cylindrical pattern.

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[CylinderRetractIssue.zip](https://github.com/user-attachments/files/16662024/CylinderRetractIssue.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-08-20:
It’s because the minimum distance for retraction is not enough to stop retractions from happening. In the latest nightlies the seam in a concentric pattern is now staggered to prevent over extrusion in a single area (if the seam was aligned).

 So the start and end points are a bit further apart than they used to be. Increase your minimum distance under the extruder settings and retraction should stop. However be mindful that the hotend needs to control flow well so it doesn’t ooze due to the larger distance.

skl111, 2024-08-23:
> It’s because the minimum distance for retraction is not enough to stop retractions from happening. In the latest nightlies the seam in a concentric pattern is now staggered to prevent over extrusion [...]

I will give that a shot. I thought I was going crazy when this started.

skl111, 2024-08-29:
So this happens quite a bit and I was able to address it using filament retraction travel distance.

I am wondering what was it before the change and what behavior was seen that required it to change to staggered?

skl111, 2024-10-10:
If there is a way to modify the travel distance for the 1st layer only, that would allow me to get the functionality back?

Or maybe fork and keep a separate build that reverts concentric pattern seam change.

I currently do not know what would be needed for that.

closed 2025-01-17 by github-actions[bot]

skl111, 2025-03-17:
I have made do with using this in the filament and extruder settings, however I just need it for the 1st layer.

Is there currently a way to make the distance apply only to the 1st layer and then revert back on the 2nd layer?

If not, I will make a feature request, as this method leaves ugly spiral scars.