OrcaSlicer/OrcaSlicer #6514: Cloning an object causes all objects on all build plates to auto-arrange

brjohnso1, 2024-08-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2dev

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5

### Additional system information

Intel

### Printer

Bambu P1S

### How to reproduce

1. Create Multiple build plates.
2. Place two objects on each plate
3.  right click any object and select clone
4. enter any number greater than zero

### Actual results

All objects on all build plates are auto arranged after the object is cloned.

### Expected results

All objects should not be arranged, especially on other build plates.
[EtsyEscapeStuff.3mf.zip](https://github.com/user-attachments/files/16666379/EtsyEscapeStuff.3mf.zip)

### Project file & Debug log uploads

[debug_Mon_Aug_19_07_32_11_34217.log.0 copy.zip](https://github.com/user-attachments/files/16666452/debug_Mon_Aug_19_07_32_11_34217.log.0.copy.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

skl111, 2024-08-20:
[image]
Here is where you can turn it off, not sure if this is meant to be on or off by default

discip, 2024-08-20:
@brjohnso1
This is already addressed with #6373.

discip, 2024-08-20:
@brjohnso1
This is already being addressed with #6373.

jwvanderbeck, 2024-10-27:
Thank you for pointing out that it is a setting but honestly this is a rather strange default setting.  I was going NUTS and had to go to the internet to try and find out why I couldn't do a simple clone.

Seems like this should default to OFF, but maybe that's just me.

closed 2025-02-02 by github-actions[bot]