OrcaSlicer/OrcaSlicer #6515: Address boundary error

Subbok, 2024-08-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1.1-1

### Operating System (OS)

Linux

### OS Version

Arch LInux

### Additional system information

CPU: AMD Ryzen 5 5600X 
GPU: NVIDIA GeForce RTX 3060 Ti
Memory: 16 Gib

### Printer

Ender 3 v2 

### How to reproduce

Try to run Appimage or compiled bin from AUR (orca-slicer-unstable-bin, orca-slicer-bin). Version 2.0.0-3 from AUR (orca-slicer) works.

### Actual results

The apps crash with this output in terminal:
[2024-08-20 00:36:04.771441] [0x00007863d7735c00] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
fish: Job 1, 'orca-slicer' terminated by signal SIGSEGV (Address boundary error)

[image]

### Expected results

Orca should open adn work. The Device tab is broaken but thats nothing new

### Project file & Debug log uploads

There's no log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

OofYeetMcGee, 2024-08-20:
Same issue

olekolek1000, 2024-08-20:
Can reproduce.
Stacktrace:
```
#0  0x00005555591846b4 in tt_face_get_paint_layers ()
#1  0x00007ffff716778d in ??? () at /usr/lib/libcairo.so.2
#2  0x00007ffff711dd00 in ??? () at /usr/lib/libcairo.so.2
#3  0x00007ffff711df87 in cairo_scaled_font_glyph_extents () at /usr/lib/libcairo.so.2
#4  0x00007ffff7ea4242 in ??? () at /usr/lib/libpangocairo-1.0.so.0
#5  0x00007ffff79d1779 in pango_glyph_string_extents_range () at /usr/lib/libpango-1.0.so.0
#6  0x00007ffff79ea01a in ??? () at /usr/lib/libpango-1.0.so.0
#7  0x00007ffff79eb253 in ??? () at /usr/lib/libpango-1.0.so.0
#8  0x00007ffff79dfb46 in ??? () at /usr/lib/libpango-1.0.so.0
#9  0x00007ffff79dff6c in ??? () at /usr/lib/libpango-1.0.so.0
#10 0x00007ffff79e026e in pango_layout_get_size () at /usr/lib/libpango-1.0.so.0
#11 0x00007ffff79ee827 in ??? () at /usr/lib/libpango-1.0.so.0
#12 0x00007ffff79e009f in ??? () at /usr/lib/libpango-1.0.so.0
#13 0x00007ffff79e41b4 in pango_layout_get_pixel_size () at /usr/lib/libpango-1.0.so.0
#14 0x00005555599640c8 in wxCairoContext::GetTextExtent(wxString const&, double*, double*, double*, double*) const ()
#15 0x0000555559706217 in wxGCDCImpl::DoGetTextExtent(wxString const&, int*, int*, int*, int*, wxFont const*) const ()
#16 0x0000555556bb1dfa in Slic3r::GUI::SplashScreen::Decorate(wxBitmap&) ()
#17 0x0000555556ba7995 in Slic3r::GUI::GUI_App::on_init_inner() ()
#18 0x0000555556ba91fa in Slic3r::GUI::GUI_App::OnInit() ()
#19 0x0000555559a13d17 in wxEntry(int&, wchar_t**) ()
#20 0x0000555556b6908b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#21 0x0000555555fc995b in Slic3r::CLI::run(int, char**) ()
#22 0x0000555555ec7539 in main ()

```

Subbok, 2024-08-20:
> Can be fixed by downgrading system `cairo` package: `1.18.0-2 => 1.17.8-2` on Arch Linux:
> 
> Install `aur/downgrade`
> 
> And then run `sudo downgrade cairo`, select `1.17.8` version.

This works, thak you very much

Maxb0tbeep, 2024-08-22:
> Can be fixed by downgrading system `cairo` package: `1.18.0-2 => 1.17.8-2` on Arch Linux:
> 
> Install `aur/downgrade`
> 
> And then run `sudo downgrade cairo`, select `1.17.8` version.

thank you

olekolek1000, 2024-09-14:
Looks like `aur/orca-slicer-unstable-bin 2.2.0.beta-1` doesn't have this issue. Can anyone confirm?

dionb, 2024-10-19:
Upgrading to 2.2.0.beta-1 worked for me. Thanks @olekolek1000 for the suggestion

gilbertorconde, 2025-01-10:
> Looks like `aur/orca-slicer-unstable-bin 2.2.0.beta-1` doesn't have this issue. Can anyone confirm?

im using `aur/orca-slicer-unstable-bin 2.2.0-1` and had to downgrade cairo. You might be talking about `aur/orca-slicer-git 2.2.0.beta2.9245a87293-1` no? (didnt test that one)

olekolek1000, 2025-01-10:
yup, I can tell `orca-slicer-unstable-bin 2.2.0-1` is broken again, interesting.
Compiled version of  aur/orca-slicer-git from scratch works. Took a little while.

closed 2025-04-19 by github-actions[bot]

BekaValentine, 2025-08-20:
I'm getting this issue on Arch with `OrcaSlicer-2.3.1-dev`. It initially worked fine, but suddenly started getting killed shortly after startup by `SIGSEGV`. Cairo downgrades don't help.

the-usr, 2026-10-31:
Just downloaded Version 2.4.0 also from `aur/orca-slicer-bin`. I am getting the same error when clicking on the Preview tab. Unfortunately i am also getting an error compiling `aur/orca-slicer-git`