OrcaSlicer/OrcaSlicer #6519: Crash when turning off advanced slider on support tab

kjcolley7, 2024-08-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

macOS Sonoma 14.6.1 (23G93)

### Additional system information

Apple M1 Pro, 32GB RAM

### Printer

Ender 3 V2 (running Klipper)

### How to reproduce

I haven't actually tried reproducing, but this is approximately what I did:

1. Have the advanced slider on
2. Go to the support tab
3. Turn the advanced slider off
4. Instant crash

### Actual results

Orca Slicer crashed with this as the report body:

[OrcaSlicer.crash.zip](https://github.com/user-attachments/files/16654876/OrcaSlicer.crash.zip)

### Expected results

No crash

### Project file & Debug log uploads

[debug_Sun_Aug_18_18_51_17_11928.log.0.zip](https://github.com/user-attachments/files/16654795/debug_Sun_Aug_18_18_51_17_11928.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2024-08-21:
Have you tried the latest nightly build?

Noisyfox, 2024-08-24:
I wonder if this is the same issue:
[image]

Noisyfox, 2024-08-24:
Could you try the build from this PR #6551 and see if that solves the problem?

kjcolley7, 2024-08-24:
I also haven't been able to reproduce the issue by doing the same steps as before. Seeing as the crash log shows that it is a heap allocation lifetime issue, it might be a sporadic crash that's tricky to reproduce. Since this crash has only happened once, it'll be tricky to see if a new build fixes the problem.

kjcolley7, 2024-08-24:
I reproduced the crash by switching to the machine g-code tab in printer settings, closing the window, and then toggling the advanced slider off (as shown in the GIF in PR #6551). This is the crash I got, which looks like it matches the crash I'm reporting in this issue:

[OrcaSlicer2.crash.zip](https://github.com/user-attachments/files/16738103/OrcaSlicer2.crash.zip)

Therefore, I assume this issue will be fixed by commit 934f171b836e5c79a30ac03a4310d58bbd380f37, once it merges. Thanks for investigating this, especially considering the lack of information I was able to provide in the issue!

kjcolley7, 2024-08-24:
After testing with the PR build (from the link here: https://github.com/SoftFever/OrcaSlicer/actions/runs/10538291424/job/29200645382?pr=6551#step:11:37), I can confirm that the crash no longer reproduces by following the steps above.

Noisyfox, 2024-08-26:
Should be fixed by #6551

closed 2024-08-26 by Noisyfox