OrcaSlicer/OrcaSlicer #6520: Orca-slicer doesn't start -  segmentation fault

kbokh, 2024-08-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1, 2.2.0

### Operating System (OS)

Linux

### OS Version

Endevour OS (Arch)

### Additional system information

CPU: i7-10510U CPU , Memory: 16Gb, VideoCard: Intel

### Printer

BambuLab X1C

### How to reproduce

1. Upgrade the system:
`[2024-08-20T22:17:03+0300] [ALPM] upgraded glibc (2.40+r16+gaa533d58ff-1 -> 2.40+r16+gaa533d58ff-2)
[2024-08-20T22:17:05+0300] [ALPM] upgraded gcc-libs (14.2.1+r26+g14fa2b2ae7f-1 -> 14.2.1+r32+geccf707e5ce-1)
[2024-08-20T22:17:05+0300] [ALPM] upgraded abseil-cpp (20240116.2-3 -> 20240722.0-1)
[2024-08-20T22:17:05+0300] [ALPM] upgraded alsa-card-profiles (1:1.2.2-1 -> 1:1.2.2-2)
[2024-08-20T22:17:05+0300] [ALPM] upgraded libsasl (2.1.28-4 -> 2.1.28-5)
[2024-08-20T22:17:05+0300] [ALPM] upgraded sqlite (3.46.0-1 -> 3.46.1-1)
[2024-08-20T22:17:06+0300] [ALPM] upgraded audit (4.0.1-3 -> 4.0.2-1)
[2024-08-20T22:17:06+0300] [ALPM] upgraded libtirpc (1.3.4-1 -> 1.3.5-1)
[2024-08-20T22:17:06+0300] [ALPM] upgraded pambase (20230918-1 -> 20230918-2)
[2024-08-20T22:17:06+0300] [ALPM] upgraded systemd-libs (256.4-1 -> 256.5-1)
[2024-08-20T22:17:06+0300] [ALPM] upgraded libx11 (1.8.9-1 -> 1.8.10-1)
[2024-08-20T22:17:06+0300] [ALPM] upgraded freetype2 (2.13.2-2 -> 2.13.3-1)
[2024-08-20T22:17:06+0300] [ALPM] upgraded hwdata (0.384-1 -> 0.385-1)
[2024-08-20T22:17:06+0300] [ALPM] upgraded kmod (32-1 -> 33-1)
[2024-08-20T22:17:06+0300] [ALPM] upgraded systemd (256.4-1 -> 256.5-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded libinput (1.26.1-1 -> 1.26.2-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded egl-wayland (3:1.1.13.1-1 -> 4:1.1.13.1-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded mesa (1:24.1.5-1 -> 1:24.1.6-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded gnutls (3.8.6-1 -> 3.8.7-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded attica (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded kconfig (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded kcoreaddons (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded kcrash (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded kdbusaddons (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:07+0300] [ALPM] upgraded libpipewire (1:1.2.2-1 -> 1:1.2.2-2)
[2024-08-20T22:17:07+0300] [ALPM] upgraded pipewire (1:1.2.2-1 -> 1:1.2.2-2)
[2024-08-20T22:17:08+0300] [ALPM] upgraded mpg123 (1.32.6-1 -> 1.32.7-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded pipewire-audio (1:1.2.2-1 -> 1:1.2.2-2)
[2024-08-20T22:17:08+0300] [ALPM] upgraded pipewire-jack (1:1.2.2-1 -> 1:1.2.2-2)
[2024-08-20T22:17:08+0300] [ALPM] upgraded libbluray (1.3.4-1 -> 1.3.4-2)
[2024-08-20T22:17:08+0300] [ALPM] upgraded librsvg (2:2.58.2-1 -> 2:2.58.3-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded libssh (0.10.6-2 -> 0.11.0-2)
[2024-08-20T22:17:08+0300] [ALPM] upgraded libdeflate (1.20-2 -> 1.21-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded libdovi (3.3.0-2 -> 3.3.1-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded ffmpeg (2:7.0.1-2 -> 2:7.0.2-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded karchive (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded kcodecs (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded ki18n (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded poppler (24.06.1-1 -> 24.08.0-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded poppler-qt6 (24.06.1-1 -> 24.08.0-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded kfilemetadata (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded kidletime (6.4.0-1 -> 6.5.0-1)
[2024-08-20T22:17:08+0300] [ALPM] upgraded kwindowsystem (6.4.0-1 -> 6.5.0-1)
[... 167 lines cut ...]
❯ ./OrcaSlicer_Linux_V2.2.0-dev.AppImage
[2024-08-21 00:21:03.937932] [0x00007b82979fdb80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
[1]    32207 segmentation fault (core dumped)  ./OrcaSlicer_Linux_V2.2.0-dev.AppImage```

### Expected results

Orca-slicer is up and running 

### Project file & Debug log uploads

No new logs are created ($HOME/.config/OrcaSlicer/log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

The same for Bambu Studio
Reinstallation has never helped

Noisyfox, 2024-08-21:
Could you provide the core dump file?

Noisyfox, 2024-08-21:
Also have you tried this https://github.com/SoftFever/OrcaSlicer/issues/6515

cyouden, 2024-08-21:
Looks like it may be an issue with another Arch package: https://aur.archlinux.org/packages/orca-slicer-bin#comment-986913

azabecki, 2024-08-21:
I can confirm this issue exists. Earlier today I was using Orca with zero issues as usual. I ran updates on my Manjaro (Arch based) Linux, approx 201 packages updated. I rebooted and Orca segfaulted. Tried the flatpack, direct download, appImage, all segfaulted. I ran:

yay -Rdd ttf-harmonyos-sans

From:

[https://aur.archlinux.org/packages/orca-slicer-bin#comment-986913](url)

It de-installed the font and Orca stopped segfaulting.

I am currently sending print jobs to my awaiting printers.....

azabecki, 2024-08-21:
> yay -Rdd ttf-harmonyos-sans

This fixed it for me.

Noisyfox, 2024-08-21:
> Looks like it may be an issue with another Arch package: https://aur.archlinux.org/packages/orca-slicer-bin#comment-986913

Good to know that, thanks!

discip, 2024-08-22:
My issue seems to be related.

I also use Linux (Manjaro) and after the last update Orca starts but no longer in dark mode. Also at least one other app (Geary [email app]) does the same thing.

Hope this helps somehow. 😊

discip, 2024-08-22:
My issue seems to be related.

I also use Linux (Manjaro) and after the last update Orca starts but no longer in dark mode. Also at least one other app (Geary [email app]) does the same thing.

Hope this helps somehow. 😊

#6515 seems to have the same cause.

trap98, 2024-08-27:
Same here, this worked `yay -Rdd ttf-harmonyos-sans`

clstr, 2024-09-14:
I was having the issue with Manjaro too and launches again after executing `yay -Rdd ttf-harmonyos-sans`, but now dark mode doesn't work.

UltraBlackLinux, 2024-09-16:
this is not a be-all-end-all solution for this problem. Apparently there's different flavors of segfaults here. 
One with fonts and one with ... No idea what

maxim-kukushkin, 2024-09-18:
Deleting the package (`ttf-harmonyos-sans`) indeed help the slicer startup. However, in my case it then always crashes with Segmentation Fault when switching between tabs in printer settings (Basic information / Extruder / Motion ability)

Not sure if it's related to I shall open a separate issue. I'm also on Arch with latest updates

Checking `strace` didn't show any suspicious output before the crash

Noisyfox, 2024-09-18:
> However, in my case it then always crashes with Segmentation Fault when switching between tabs in printer settings (Basic information / Extruder / Motion ability)

This is a known issue and should have been fixed in 2.2.0 beta.

CitrusIntellect, 2024-09-18:
You can try installing it from the AUR (orca-slicer-bin) 
just a warning, I see your using an arch based distro so try at your own risk

maxim-kukushkin, 2024-09-18:
> You can try installing it from the AUR (orca-slicer-bin) just a warning, I see your using an arch based distro so try at your own risk

I think AUR is still on 2.1.1, so I'll probably have to wait until 2.2.0 is released

clstr, 2024-09-19:
> > You can try installing it from the AUR (orca-slicer-bin) just a warning, I see your using an arch based distro so try at your own risk
> 
> I think AUR is still on 2.1.1, so I'll probably have to wait until 2.2.0 is released

Download the 2.2.0 beta appimage and run it. the one on the aur is the appimage too.

kohbo, 2024-10-29:
This issue is occurring for me on version 2.2.0-1

kohbo, 2024-10-30:
Switching to the latest version on Github 2.2.0 beta 2 is working for me

padcom, 2024-10-30:
> Switching to the latest version on Github 2.2.0 beta 2 is working for me

Tried that one, tried the Ubuntu24 one - nothing works. Only disabling networking for the printer.

su-sd, 2024-11-20:
I also have a segfault at the startup. I tried v2.2.0 and to build it myself.
I'm on debian12. Running orca with strace reveals that it always crashes after loading gstreamer stuff.
I attach a log.
[orca.txt](https://github.com/user-attachments/files/17837471/orca.txt)

su-sd, 2024-11-22:
My workaround is to use an arch linux root and start orca-slicer with bubblewrap.

fabiovix, 2024-12-04:
I have this problem too with Ubuntu 24 and Orca Slicer 2.2.0 (tested 2.3.0dev  too, with no luck).
Today, I managed to get it working:  i've done port forward (NAT) of the ports:  2222, 80, 9999, 5353, 4409, 4408 and 8080 on my firewall to access it outside my house.  Then, I configurated OrcaSlicer to the external IP and external port and, strangely enough, Orcaslicer worked with both Fluid (port 4408) and mansail (port 4409).   I hope this information can be helpful to solve the bug.

su-sd, 2025-03-05:
V2.3.0 beta starts under debian 👍

closed 2025-06-11 by github-actions[bot]