OrcaSlicer/OrcaSlicer #6521: Bed Texture Offset

ahmunro, 2024-08-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

Custom DElta

### How to reproduce

Create a circular bed of any dimension. 

### Actual results

The grey bed colour is not aligned with the bed & bed texture
[image]

### Expected results

The grey colour should align with the bed texture

### Project file & Debug log uploads

[Untitled.zip](https://github.com/user-attachments/files/16685577/Untitled.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

neil-tan, 2024-09-28:
I'm having the same issue here.

straga, 2024-12-22:
That is still in the version 2.2.0.

lukealford, 2025-02-20:
Can confirm, still a issue in v2.2.0

neil-tan, 2025-04-09:
It is still an issue in v2.3.0.

closed 2025-07-16 by github-actions[bot]