OrcaSlicer/OrcaSlicer #6523: Special characters in gcode name causes file unable to be printed

Ataraxia-Mechanica, 2024-08-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 nightly

### Operating System (OS)

Windows

### OS Version

Klipper

### Printer

All klipper machines

### How to reproduce

1. Slice a model with character "#" in it
2. Generated Gcode has "#" in its name
3. Upon uploading to the printer, "Malformed command" error happens
[image]

### Actual results

Unable to print the file

### Expected results

Slicer automatically removes the "#" symbol from the filename so it can be printed

### Project file & Debug log uploads

[Geometric Sculpture #07.zip](https://github.com/user-attachments/files/16691244/Geometric.Sculpture.07.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MiPnamic, 2024-09-21:
I was working on a prototype called Body_01, as I was at the second iteration of the same model the slicer decided to add a star (*) to the filename making it:

Body_01*_PLA_10h49m.gcode

Second iteration means:
- slicer open with the old model already printed
- back to prepare
- select and delete the model
- uploaded the new version (same 3mf filename, same object name inside)
- sliced
- upload and print

at the name suggestion I saw the start but I ignored that, then Klipper told me that a Malformed command was sent.

tlhintoq, 2024-12-15:
Kinda goes with this
https://github.com/SoftFever/OrcaSlicer/issues/7749

thebeeric, 2024-12-19:
In my case, I get no error, just a silent failure when the gcode name has an "#". Obviously, the workaround is to modify the filename, but it wasn't clear why this specific print wasn't starting. My experience:

1. Work on model with name "something#1".
2. Try to print from Orca with the default name
3. Orca attempts to transfer "something#1.gcode" to my printer
4. Transfer completes without errors, but the file cannot be found in the printer and the print never starts

If you try to print with special characters in the filename, Orca should prevent or at least warn the user special characters shouldn't be used.

tlhintoq, 2024-12-20:
Sounds like a single routine for parsing all filenames down to a lowest-common-acceptable level could be used to handle both of these, and do so consistently so it doesn't become a case of
> I can use this here, but not here.  Over here blahblah is okay, but not here

Ataraxia-Mechanica, 2025-06-20:
bruh

closed 2025-09-27 by github-actions[bot]