OrcaSlicer/OrcaSlicer #6529: Top friction layer

rkarlsba, 2024-08-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It's not really related to a problem I know of, except that fuzzy skin doesn't work on the flat top of a model.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

For some applications, it'd be nice to add a last layer (or two) on the top to rough it off, perhaps like fuzzy skin, but for the top (and potentially also bottom) when these are flat. This should be possible by just adding a separate layer after the solid ones just to add structure, for friction or different reflection. I guess this could be done with just adding another layer using an infill, typically with a high percentage.

Would this be an easy task? I'd love to see it :)

### Describe alternatives you've considered

1. fuzzy skin
2. removing the top layer, but then, this makes all infill very dense and is very slow

vgdh, 2024-08-22:
I think it can be done with just moving the nozzle little up and down during the infill procedure.
It would be pretty cool.

rkarlsba, 2024-11-21:
It still would be a cool feature, so please keep this open

closed 2025-02-28 by github-actions[bot]