OrcaSlicer/OrcaSlicer #6534: Website download links not quite right.

craig9, 2024-08-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Linux

### OS Version

N/A

### Additional system information

N/A

### Printer

N/A

### How to reproduce

Go to website download page, click on linux download.

### Actual results

Mac dmg file is downloaded

### Expected results

Linux AppImage is downloaded

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

The links appear to be reversed, and also they are not up to date with the current 2.1.1 release.

Might it be better (less maintenance) to point website visitors to the releases page on github which is up to date?

LazyCoderOZ, 2024-08-23:
I think craig9 means this one https://orcaslicer.net/#download-orca-slicer
The link to the Linux Version download is actually https://github.com/SoftFever/OrcaSlicer/releases/download/v2.1.0/OrcaSlicer_Mac_arm64_V2.1.0.dmg

The MAC version is actually the Linux version https://github.com/SoftFever/OrcaSlicer/releases/download/v2.1.0/OrcaSlicer_Linux_V2.1.0.AppImage

It appears the files have been associated with the buttons incorrectly.

craig9, 2024-08-23:
Yes - as @LazyCoderOZ mentioned the mac and linux links on that page are reversed. And the linux link is out of date, at least as compared to the releases available on github.

@ElectricalBoy apologies - I (perhaps mistakenly) assumed there was only one orcaslicer website. I should have specified.

craig9, 2024-08-23:
@ElectricalBoy that is strange, orcaslicer.net is the top link when searching the web for orcaslicer, it certainly looks like an official site. The Ko-fi and discord links appear to be genuine, and are hosted by accounts also named SoftFever.

I initially reported the bad links issue in the official discord group, but was re-directed to report it here at github.

What would you say is the next step?

thejudge156, 2024-09-02:
Howdy, Questcraft Lead Developer here. 

We also run into this exact format of site (questcraft.net), which is not hosted by us. A DMCA request to the hosting provider clears it for a few months, but it always reappears. Just thought you should be aware.

Merikrotti, 2024-09-24:
Adding here, I installed the OrcaSlicer from there aswell. The update manager brought me here though. The original install file matches the hash that is available here, so I'm probably not infected. That might change soon, now that they have the number 1 spot on search results and quite convincing site.

tavdog, 2024-10-02:
Yeah it's a pity this false site is the top result in google.

closed 2025-01-09 by github-actions[bot]

MrSuicideParrot, 2025-01-27:
Only for information, it appears there is a new malware site: https://orcaslicers.com/.