OrcaSlicer/OrcaSlicer #6544: manual control of maximum volumetric speed for external walls

BlasterGales, 2024-08-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The exterior finish of prints with certain filaments is usually irregular between glossy or matte due to the variation in the extrusion speed. In the corners it is usually brighter due to the reduction in speed, on the contrary in the straight areas it is usually more matte due to the increase in speed. This is partially solvable by slowing down the speed A LOT (The finish will have maximum gloss depending on the temperature), but it is not the root of the problem, it is just a temporary solution.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

add a manual control of maximum volumetric speed for external walls, this control will prevent the material flow from varying during the printing of the visible walls, causing the plastic to melt at the same speed, giving a uniform finish, whether glossy or matte, and not a variation of both. This will allow the advantages of high acceleration to be maintained as long as it does not affect the flow of the material. On the other hand, it is possible that this may have an improvement in strength for prints that require it, but I have not investigated this.

### Additional context

This control exists in the filament settings, but its use is intended to work within the limits of the material. To achieve a uniform finish on prints, some filaments require lowering this setting to 3mm/s and doing so adds a lot of printing time because it is used for the entire print (infill, internal walls, supports, etc.). Adding this option in the print settings in the "quality" tab only for external walls would be ideal, making this maximum speed setting override the filament setting only during printing of the external walls.

I feel that not enough importance is given to the extrusion speed in printing programs, it is an important characteristic in the formation and unification of plastic, being also a critical point to configure in filament recycling extruders.

gueee, 2024-08-24:
this phenomenon can be tackled in a different way, since the root cause is temperature saturation in the filament, caused by how long it sits in the melt zone of the hotend.
there's already a project on github that addresses that exact issue with post processing the g-code to vary the hotend temperature according the flow rate.

here's the link to that project if you're interested.
https://github.com/sb53systems/G-Code-Flow-Temperature-Controller

avogra, 2024-09-18:
Why don't you adjust the print speed for outer walls in the speeds tab? I don't see any use in having both, print speed and max flow for outer walls in the process settings.

In general, matte surfaces of otherwise shiny filaments indicate insufficient melting, which causes not only changes in the surface but also e.g. weak layer adhesion. So I typically not only reduce speed or flow of the outer walls but also of all other line types, because I don't want hidden flaws inside the model. In that light, a single parameter for max flow max perfect sense.

bo0tzz, 2025-02-15:
This is not stale

nekenyu, 2026-08-19:
Another reason for max volumetric flow for external walls: External Walls have a different user-defined tolerance for defects than the internal printing.

My recent calibrations have had a point of line failure and a lower point of minor defects that I would rather not have on the surface. These give two often different max speeds.