OrcaSlicer/OrcaSlicer #6558: infill multiplier

carrierwave42, 2024-08-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I can only do single wall infill in Orca. I can use infill multiplier in Cura to create stronger multi-wall infill patterns. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be amazing if Orca could implement multi-wall infill like Cura. I found this quite useful at times in Cura. It's called infill line multiplier in Cura.

### Additional context

[image]

kisslorand, 2024-09-06:
> ### Is there an existing issue for this feature request?
> * [x]  I have searched the existing issues

https://github.com/SoftFever/OrcaSlicer/issues/3731

carrierwave42, 2024-09-08:
Yep, GitHub UI is awful.
I found this after submitting my issue. Slightly different search terms… 
Anyway same issue, please promote it.

—
Mark Gunning
***@***.***
gunningdesign.com (web)
+61 457 523 463 (Mobile)
ahhadesign (skype)

closed 2024-12-15 by github-actions[bot]