OrcaSlicer/OrcaSlicer #6560: Step file wont be rendered correctly in orca/bambuslicer

marcosikkens, 2024-08-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Ryzen 9750X 6700xt

### Printer

Bambulab x1c

### How to reproduce

I have made the following model as a step file:
[image]

Exported it in Onshape as a .step file. and loaded it in orcaslicer. Which gives the following rendering:

[image]

### Actual results

[image]

It looks like the top sphere is missing, It is still not there after slicing. However after exporting it as an STL it does work. bambuslicer behaves exactly the same.

### Expected results

The top sphere should be rendered correctly

### Project file & Debug log uploads

The project file can be found below:

[Part Studio 1 - Part 1(24).zip](https://github.com/user-attachments/files/16739953/Part.Studio.1.-.Part.1.24.zip)

The step file is added as a zip also.

[Part Studio 1 - Part 1(2)-step.zip](https://github.com/user-attachments/files/16740072/Part.Studio.1.-.Part.1.2.-step.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

TobiLaForge, 2024-08-26:
That seems to be a problem with your step file, not with any of those slicers. I opened the step file with CAD Assistant and with FreeCAD and they also don't show the sphere (only a curved top surface).

marcosikkens, 2024-08-26:
I loaded the step file in the autocad online viewer and it shows correctly. I am now confused that it doesn't for you.

foreachthing, 2024-08-27:
Your STEP is 'STEP AP242'... when I save the same STEP as 'STEP AP214' (via SolidWorks 2020), OrcaSlicer opens it as expected.
So, try to find a export setting for STEP files and change the AP version - if possible.

marcosikkens, 2024-09-02:
That linked issue seems like the same problem and is also better described. 

There were some discussions in 2018 about moving to ap242 because it had some advantages as can be seen in the post here

https://forum.onshape.com/discussion/7866/step-version-on-export-ap214-242

closed 2024-09-02 by marcosikkens