OrcaSlicer/OrcaSlicer #6562: Individual compensations  X,Y,Z

boromyr, 2024-08-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be more useful and versatile to be able to compensate the individual axis instead of the XY shrinkage, adding a scale multiplier for each axis. This would simplify the calibration processes for each printer without modifying the firmware or steps per mm, or alternatively it can also be used to compensate for shrinkage.

### Describe alternatives you've considered

All calibration processes

idkjustgoogleit, 2024-08-26:
+1

at least some form of z-height shrinkage compensation instead playing with top shell thickness settings, or scaling each print manually...

boromyr, 2024-09-02:
In the dev version I saw the compensation of the Z axis has been introduced, it should be quite simple therefore separating xy competition

Sidiusz, 2024-09-05:
That's a good suggestion

boromyr, 2024-12-06:
Up

DusekM98, 2025-02-11:
I agree. I don't see a reason why it shouldn't be there. At least in the advanced options. Compensation is just a scale operation, so splitting it into X and Y can't be that difficult. It would also improve the accuracy due to shrinkage differences in X and Y.

cclix, 2025-04-14:
+1

Would be useful on the Kobra 3, because as for now you can't set the Rotation Distance in the printer.cfg cause the firmware is locked up. Would be possible with custom firmware but not with stock. 🙄

boromyr, 2025-09-14:
You need to manually set the scale for each model if you want to apply different retractions or dimensional corrections to compensate for small defects of your printer. Having them independent at the slicer level would mean greater freedom for anyone: you can use them to compensate for material shrinkage, or even to calibrate the step/mm without modifying firmware/.cfg not only at the machine level but also for each filament type.  
No printer supports modifying the step/mm per filament type, only globally; this way you can dimensionally calibrate your printer for any spool you have, or perhaps for other uses we haven't imagined yet.

boromyr, 2025-09-16:
With separable variables, you can calibrate the steps/mm more easily; in fact, there's no need to calibrate the steps or the shrinkage anymore, you can calibrate both in a single step. The common offset of X and Y from 100% will correspond to the shrinkage, the discrepancy between the two will be the steps compensation. Then, anyone who wants can always calibrate both things in the traditional way.

boromyr, 2025-09-16:
It doesn't matter what you call it, the effect is the same; it's up to the user to decide how to use it.  
No proof is needed; it's mathematically elementary: if you scale a model by 200% or compensate the steps for mm to print 200% of the normal size, in both cases you'll get a model twice the original size. However, one thing that the step compensation in X or Y doesn't account for is that the extruder doesn't see the compensation of the XY motors, resulting in over-extrusion or under-extrusion. Consequently, a model printed with motors compensated at 200% will be under-extruded by up to 50% on both axes. A mass of filament creates a ghostly shape. And the motors are stressed twice as much.  
This is an exaggerated case to better explain the concept I want to convey.  

Therefore, the XYZ scaling in the slicer is better because:  

- You can use it to compensate for XY and Z shrinkage  
- You can use it to compensate for X, Y, and Z steps  
- The compensation does not affect extrusion because the slicer performs the correct calculations on the new dimensions  
- It does not require firmware modification, which is better for everyone, especially for those who do not have access or are less experienced  
- You can always choose not to use any of the above and stick with the current mode