OrcaSlicer/OrcaSlicer #6564: Slow performance on large models

Cagle1991, 2024-08-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When slicing large or very large models the slicer takes a long time 15-20 minutes especially with supports added.  It seems that the program is only utilizing one core.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Is it possible to incorporate more cores or a dedicated gpu to help mitigate the load
[image]
 while slicing?

tlhintoq, 2024-08-27:
X1C ?  The printer is too small to even fit a large object.

Did you notice the warning that tells you that the object has a massive number of triangles **_and thus processing it will be slow_** in its mesh and that you can simplify the model (generally with zero affect on its appearance) ?
FDM at this physical size for what looks effectively like little more than a helmet dome won't show any difference if you simply this dramatically.

tlhintoq, 2024-08-27:
On this other slightly similar ticket @Noisyfox mentions that using the `Tree: Strong` support combination is not able to be parrallelized.  @Cagle1991 maybe that's happening here: The support type you're using is the cause of the slow down.  Costs ya nothing to try another type and see if processing picks up.
https://github.com/SoftFever/OrcaSlicer/issues/6098#issuecomment-2312484259

closed 2024-12-03 by github-actions[bot]