OrcaSlicer/OrcaSlicer #6572: Innovative Non-Intersecting Honeycomb Infill - 31% Faster

Ergonomicmike, 2024-08-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No. It's simply a faster Honeycomb infill.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

This request would integrate[ m_IDEX's Innovative Non-Intersecting Honeycomb Infill](https://www.printables.com/model/986103-innova) into Orca, by automagically changing some settings in Process when this Non-Intersecting Honeycomb infill is activated. (I suggest that, as with settings like Spiral mode, Orca should tell the user that certain settings will be changed, and "Do you want me to change them for you?" so that the user can know what to undo if they undecide to use this new infill.)

### Describe alternatives you've considered

Do nothing. Maybe the world doesn't need a faster Honeycomb infill?

### Additional context

This is not my idea. I saw it on printables. I presume that the author could create his own Feature Request here. But perhaps he's too humble to "toot his own horn."

Ataraxia-Mechanica, 2024-08-27:
Wouldn't this be significantly weaker than double wall honeycomb as a third of the walls are printed in the air? It's also about 50% slower than double wall honeycomb, as single wall = smaller hexagon size to fill the same portion of space as double wall honeycomb, needing more turns in the same amount of space.

Every speed value except travel including bridging speed is set to 275mm/s and 17500mm^2/s here, note the same weight of material used:
[image]
[image]

Ergonomicmike, 2024-08-27:
Interesting. Since I didn't try it, I don't know about speed/usage. But your screenshots tell the tale. So maybe this is not a good idea.

As for wearing down an extruder due to flow rate changes, I don't know. I don't understand why asking an extruder to operate over its range would cause it to wear out faster. 

I can maybe see how a lot of back and forth Pressure Advance/Retraction moves might cause more gear wear due to diametric momentum changes. But in the scheme of things, even if that caused more wear, would it be noticeable over the life of a printer?

Ataraxia-Mechanica, 2024-08-27:
I've used very cheap HGX extruder before, and didn't enable travel distance threshold so every print move is ended with a retraction, and used very high retraction speed. And just after 3 months of use, the bearing completely wore off, so it's definitely noticeable.

Ergonomicmike, 2024-08-27:
Okay. I haven't printed enough to accrue time for a failure. 

I have chewed through filament when my travel distance threshold was set too low. 

Still, I've never taken flow rate changes into account when I slice a print. I've used about six spools in the year and a half that I've been printing. Have only noticed the slightest of wear on my cheap Creality extruder gear.

In any event, it's looking like this new Honeycomb infill isn't such a good idea.

Thanks for testing it.

dromanovich, 2024-08-27:
Thank you for looking into this idea.  The speed gain is due to effectively lower infill density to achieve the same size hex grid.  The pattern that I designed in CAD had approximately 13 hexagons across.  To achieve the same cell count on the same size cylinder, I used 16% honeycomb infill.

Note that this is not about print speed, but rather a perfect single wall honeycomb infill for cosmetic purposes.

DVSVIDEO, 2024-09-03:
This is not a new idea. I used to use the Simplify 3D slicer. There it was used as Fast Honeycomb years ago.
https://www.simplify3d.com/resources/videos/infill-pattern-basics/
It was really faster than the regular Honeycomb and the lines didn't cross each other on the same layer.

slothy89, 2024-09-07:
> Wouldn't this wear down the extruder as it requires doubling and halving flow frequently, or needing to bridge, or print a third of the infill at half the speed? It's also about 29% slower than double wall honeycomb, as single wall = smaller hexagon size to fill the same portion of space as double wall honeycomb, needing more turns in the same amount of space.

I feel this could be handled in the same way that CrossHatch, Gyroid, and 3D Honeycomb handle the transition layers where the infill swaps direction by 90 degrees. No need for flow or move speed changes, it's infill, not walls. Even Rectilinear and Line infills do this without incident.

This seems like a good option to add.

closed 2024-12-15 by github-actions[bot]

d0u8l3m, 2025-02-10:
Can we reopen this and take another look at having these settings as a preset? thank you

314159-r, 2025-09-29:
This elegant pattern can be simplified by printing the entire line with a constant flow rate of 1.5. Eliminating any need for repeated change of flow rates. Except for some over-extrusion on the first layer (which could be handled), it will give an identical result.

I would expect this pattern to have a substantial improvement in compression strength to weight ratio over the conventional honeycomb pattern. Possibly even more than 33%, as the alternating dual/single wall layers might loose strength, where the single line could act as a wedge, splitting the dual lines. It would be very interesting if someone with measuring capabilities could perform a comparison test...

I don't think the comparison by [Ataraxia-Mechanica](https://github.com/Ataraxia-Mechanica) is quite fair, considering the big difference in cell size. To me, infill density is mainly about saving plastic while preserving most of the strength. Not just to save time or reach a certain object weight. 
Take a look at the two comparison images by `Ataraxia-Mechanica` at the top, and tell me which one you believe would be the strongest and take the most load..? 
(both have the same weight)

I think this pattern has great potential. It appears easy to implement, reasonably fast to print, and has possibly a great strength to weight ratio. What else can you ask for..? Oh yes, as [dromanovich](https://github.com/dromanovich) said: it looks good when printed as a grid.