OrcaSlicer/OrcaSlicer #6575: Option to include 'large' flat surfaces in 'topmost' ironing setting

TrentBartlem, 2024-08-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have a print with large, flat surfaces that also hasdetails that I don't want ironed (stuff that's only 1 layer wide, thin recesses that are flat but not really visible etc).

With All Top Surfaces ironing, it goes overboard, trying to iron away the details:

[image]

but with Topmost ironing, it neglects the obvious flat surfaces that stick out slightly less:

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The Ironing section  should provide a checkbox to enable this setting, and a number input box  to determine a radius.

If it is all-surfaces, then it will *exclude* surfaces below that radius

if it is topmost, then it will *include* other surfaces it sees

### Describe alternatives you've considered

1. point-and-click manual ironing, like supports - nice, powerful, but may be more effort and/or require users to do more manual steps. I guess you could combine the features, like auto-vs-manual supports

TrentBartlem, 2024-08-27:
It's possible (or likely) that a 'select and fill' approach may be the most effective in the end.

kylek29, 2024-08-28:
> It's possible (or likely) that a 'select and fill' approach may be the most effective in the end.

That's what I was thinking, like the painting tools for supports/seams but for ironing.

I run into issues when ironing with all top layers and thin text, so it would be nice to have it get some love. It's a cool feature.

closed 2025-03-05 by github-actions[bot]