OrcaSlicer/OrcaSlicer #6576: Texture placement doesn't match the bed model, previous version 1.9.0 made more sense

SO-cyk, 2024-08-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

no

### Printer

Creality K1/ InfiMech TX

### How to reproduce

In the previous version 1.9.0, textures could be displayed on the outside of the platform area, matching the manufacturer's stl model.
Whereas in version 2.1.1, textures can only be displayed on the inside of the platform area, offset from the factory stl model. The display is not as good as it could be
[image]
[image]
[image]
[image]

### Actual results

The display can be optimised to be more aesthetically pleasing. Conformity with reality

### Expected results

The display can be optimised to be more aesthetically pleasing. Conformity with reality

### Project file & Debug log uploads

[Untitled.zip](https://github.com/user-attachments/files/16758898/Untitled.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Hammerfest, 2024-08-27:
"more aesthetically pleasing"

If your aligning to grid lines, its more then just aesthetically pleasing, that is actually kinda nuts, ive kept my profiles from 1.8 or 1.9, ill have to force clear to see this to find out if my printers are impacted as well

ilko-k, 2024-09-15:
same as https://github.com/SoftFever/OrcaSlicer/issues/5883 ?

SO-cyk, 2024-09-18:
> same as #5883 ?

Related to this issue, this issue all feeds back that the bed model and textures are shifted.
What I'm trying to feed back is that the logic for texture placement is glitching, and through observational testing the current version of the textures only show up within the print platform area. Although this does not affect slicing, just aesthetics.

closed 2024-12-26 by github-actions[bot]