OrcaSlicer/OrcaSlicer #6583: Slow down for overhangs + Arachne causing wall artifacts

aartrost, 2024-08-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

OSX 14.4.1

### Printer

Voron 2.4

### How to reproduce

1. Open this STL: https://www.dropbox.com/scl/fi/v69swdh60a5isa5dm0uc1/orca-e-step-bug.stl?rlkey=ai8nfks83fboea0a0b3olhpcj&st=7x8o4koq&dl=1
2. Slice with default Voron profile (I think this doesn't matter too much, as long as Arachne and slow down for overhangs are enabled)
3. Check the Gcode output for the outer walls and notice the E-steps being wildly varied
[image]
4. Set Wall Generator to "Classic" and enable Overhang Speed "Classic mode" and notice the E-steps being stable
[image]

### Actual results

[image]

### Expected results

See above

### Project file & Debug log uploads

[orca-e-step-bug.3mf.zip](https://github.com/user-attachments/files/16768069/orca-e-step-bug.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TinyMito, 2024-08-28:
fyi Bambu just released a fix for this: https://github.com/bambulab/BambuStudio/releases/tag/v01.09.05.51

aartrost, 2024-08-28:
> fyi Bambu just released a fix for this: https://github.com/bambulab/BambuStudio/releases/tag/v01.09.05.51

Well shit, glad to see I ran into an actual issue. Pulled my hair out diagnosing this one. Thanks for the heads up!!

igiannakas, 2024-08-28:
Try the latest nightly. There have been some improvements in the overhang segmentation logic that may be related to what you’re seeing.

aartrost, 2024-08-28:
Just gave the latest nightly a try and that didn't seem to fix it unfortunately.

igiannakas, 2024-08-28:
BTW the fix from bambu above is not related to the issue you're facing. 

More likely it has to do with the SCV value that is set for the model. I believe what you're seeing looks like this:

[image]

From Ellis's print guide: 

- Your square corner velocity or jerk may be too low (or too high!)
- This can also be a sign that your perimeter speeds/accels are too high.
- If you are using input shaper, try disabling or re-tuning it.
- Check over your printer axes for mechanical issues. Move the toolhead/bed by hand and see if anything feels off.

https://ellis3dp.com/Print-Tuning-Guide/articles/troubleshooting/bulging.html

When printing with classic mode + classic overhangs your overhang print speed is 44mm/sec
[image]

When printing with Arachne + classic disabled, your overhang print speed is 50mm/sec
[image]

My gut is telling me that this may be hitting a resonant frequency in the motion system. 

You could try these settings with **Arachne enabled** which would print the overhangs at the same speed as the classic + classic version above:
[image]

igiannakas, 2024-08-28:
BTW the latest nightly cleans up those tiny speed transitions. This is with your model file from above, with no changes made:

[image]

So if the nightly has not fixed this, it is not because of the tiny speed adjustments.

aartrost, 2024-08-28:
Thank you for the super detailed reply! I'm heading out for a trip right now and I'll pick this back up when I get back to see if any of your suggestions do the trick.

aartrost, 2024-09-03:
Hey!

I just got back from the trip and tried out the latest nightly again. With the overhang speed at 44mm/s I still see the same behavior. If you check out bottom of the print you can see a smooth line around the layer seam, then stuttering right after (I have input shaper disabled for these prints FYI, with it enabled the printer mostly smooths out the issue).

[image]

When looking at the slicer you can see that the gcode steps correspond with what I'm seeing on the print, the first few G1 commands all move ~0.5mm and extrude ~0.02 then the next G1 has only 0.002 Y movement and an E value of 0.00006, a 300x reduction. Could this be what is causing the printer to stutter?

[image]

I've followed Ellis's print guide and I think my printer is mechanically sound, earlier I printed this complicated ERCF part without any defects:  

[image]

igiannakas, 2024-09-03:
It may - maybe try reducing the slicing resolution? Is the model an stl or step file?

aartrost, 2024-09-03:
The model is an STL coming from this popular engine printable: https://www.printables.com/model/572421-3d-printable-jet-engine/files#preview:file-AjnXt. I used Orcaslicer's Cut tool to cut out the affected section for convenient reproduction of the issue.

When reducing the slicing resolution from `0.012` to `0.02` I definitely see some improvement in the gcode. Looks like there's 2 microstutters at 90 degrees from the seam and some ringing after. 

[image]

[image]

At `0.03` the stutters are fully eliminated but obviously at a cost in accuracy:

[image]

[image]

I've also tried to reproduce the issue in Prusaslicer but the model prints beautifully here. No ripples whatsoever and the G1 steps all look very smooth.

[image]

[image]

igiannakas, 2024-09-03:
I’ll give it a test print when I’m back on Sunday - do remind me if you can though :) 

These micro segments shouldn’t really cause an issue but let’s check

aartrost, 2024-09-03:
Much appreciated! 

I could definitely see how it could be my printer responding too aggressively to the sudden change in step size. A/B testing the issue on a separate printer would be a great way to get some insight into that. I'll set a reminder to check in on Sunday, thanks again.

aartrost, 2024-09-08:
Hey Ioannis! If you're still up for it, here's the requested reminder to see if you're able to reproduce the issue from your end as well. For the tests above I have `input_shaper` disabled to accentuate the behavior. With `input_shaper` enabled it's harder to reproduce.

igiannakas, 2024-09-10:
Hey! Sorry for the delay! Just got around to printing it. 

**Test 1: My regular print profile:**
I used my regular printing profile and the latest nightly in Orca (2.2 beta) to run it through. I've attached it below for your reference.

I don't see the issues you've faced on the print - surfaces look very smooth. I am using Arachne for the print plus slowdown for overhangs and a 0.0063 slicing resolution. Slight over extrusion on the top surface in my print, as my filament is a bit wet from not being used for a month or so :)

[image]
[image]

Attached is my project file. If you are to use it, make sure you replace your start print/end print gcodes and filament etc settings with yours.

[orca-e-step-bug - IG test print.3mf.zip](https://github.com/user-attachments/files/16942379/orca-e-step-bug.-.IG.test.print.3mf.zip)

**Test 2: Standard Orca Voron 0.2 profile**
I've also run a test print with the default voron 0.2 profile. Project file below.
[orca-e-step-bug - IG test print - default voron profile.3mf.zip](https://github.com/user-attachments/files/16942454/orca-e-step-bug.-.IG.test.print.-.default.voron.profile.3mf.zip)

Used grey filament this time to see better if there are any issues.

Here I could see the problem. Also I noticed that in this Gcode, the micro extrusions do happen, vs in my standard profile that they dont.

[image]

So your assumption that these micro extrusions are to blame is correct.

[image]

**Investigation - Line widths:**
**Increasing the external wall line width to 0.42 in the default voron profile at 0.2 makes these micro segments disappear** - see below

[image]

[image]

And the print looks good. So indeed it is those micro extrusions.
 
**Investigation - Overhangs:**
Disabling slowdown for curled perimeters doesnt remove these micro segments 

[image]

**Disabling slowdown for overhangs fully removes this issue:**
[image]

So I am inclined to believe that the line segmentation algorithm is to blame here. The fix introduced with this pr: https://github.com/SoftFever/OrcaSlicer/pull/5996 while it addresses the speed change issue that occurred in 2.1.1 doesnt fully address the generation of micro segments when they are not needed.

@SoftFever  I'll try to introduce a filter in the std::vector<ExtendedPoint> estimate_points_properties of the extrusion processor to filter out these micro segments.

igiannakas, 2024-09-10:
So I've got a potential fix ready - the below is with the standard voron profile that was causing the issue above.
[image]

Needs a bit more testing before I push out a PR for this.

igiannakas, 2024-09-10:
I've pushed a PR for testing. The build can be found here:
https://github.com/SoftFever/OrcaSlicer/pull/6714

Checks -> build all -> download the appropriate build file.

aartrost, 2024-09-10:
Thank you so much Ioannis! I'm really grateful you went the extra mile to reproduce this. Cheers and have a great rest of your week!

igiannakas, 2024-09-10:
No worries Aart! It was a tricky bug, and one that fixing it should benefit print quality for the slicer so it grabbed my attention 😀 

do test out the latest nightly when it’s compiled, it has been merged in by soft fever. Will need users to confirm that the fix is working and you’re the best person to do so 😀

igiannakas, 2024-09-10:
I saw the same on mine. Try increasing the external wall line width to 0.44-0.46 and the overhang will print much better

aartrost, 2024-09-10:
Woah crazy, that made a huge difference. Did not think to look there for better overhang performance. Thanks for the heads up

igiannakas, 2024-09-10:
> Woah crazy, that made a huge difference. Did not think to look there for better overhang performance. Thanks for the heads up

Have a look at the wiki - slowly we're putting some more info there. You'll find that tip on line width in the page below ;):
https://github.com/SoftFever/OrcaSlicer/wiki/quality_settings_line_width

Wiki homepage:
https://github.com/SoftFever/OrcaSlicer/wiki

igiannakas, 2024-09-10:
Also try one wall top and bottom surface and monotonic line for the top and bottom surfaces. That will make that first layer gap disappear ;)

aartrost, 2024-09-10:
Excellent, I've adjusted my line widths according. I've tweaked almost all settings in my custom profile but for some reason never thought line widths would be very consequential.

PR #6714 (2024-09-10): Filter out micro segments generated during the overhang slowdown process
# Description

During the overhang slowdown process, the overhang lines are split to apply the slowdown algorithm.

However there are certain cases where micro segments are generated (less than 0.01 mm in length). These micro segments result in wall extrusion consistency issues as reported here: https://github.com/SoftFever/OrcaSlicer/issues/6583

This PR aims to filter out the generation of these micro paths by ensuring that any new points generated are at least 25% of the line width long.

Fixes #6583

# Screenshots/Rec [... cut]