OrcaSlicer/OrcaSlicer #6584: Send print job to preview wrong colors

f-ij, 2024-08-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

I have an A1 printer, synced with the AMS. I select the colors for my model in the slicer using the colors from the AMS. Then when I want to print, the preview shows up prompting to select the colors in the AMS to correspond to the colors selected in the slicer. They are automatically selected to be the right ones.

### Actual results

The 3D preview of the print shows the right colors for a spit second, then permutes the colors. The print print normally, and not with the colors shown in the end in the preview. The preview on my printer shows the correct colors.

### Expected results

The right colors should be showing.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16768395/log.zip)
[LianCoinBlue.zip](https://github.com/user-attachments/files/16768403/LianCoinBlue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

LouisH98, 2024-08-28:
I am also seeing this! Even on projects that have a single colour, it somehow still shows the wrong colour in the preview box

Keno55, 2024-08-28:
Same here. The Preview shows X+1 color. Choosen color 2, preview 3

brjohnso1, 2024-09-16:
Same. Been doing this for quite awhile now. In addition, Orca selects the wrong AMS slot when you don't have spools in every slot, and you select a slot after one that is empty. For example, If I have spools in slots 1, 3, and 4, and I set the model to print from the filament in slot 3, Orca selects AMS slot 1, at least that's what shows in preview. The only fix is to remove the filament in the first slot, or move the one in slot 3 to slot 2, and re-slice.

richard-scott, 2024-09-30:
I was going to create a new issue about this on 2.2.0-beta2:

[image]

LeoStreit, 2024-10-13:
Same her!

[image]

bergie5737, 2024-10-28:
I have the same problem. Yesterday I selected green filament in slot 3. The preview showed white, (there was white in slot 1) It printed in white. :-(  It may have been a mistake on my end, but it was a reprint, so not sure why it used slot 1 in this case. What I want to say, I was used to seeing the wrong color, so did not bother to check the actual slot number.

closed 2025-02-03 by github-actions[bot]