OrcaSlicer/OrcaSlicer #6585: P1S Device Camera Crashing Slicer (Linux, Bazzite)

foxhull, 2024-08-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2dev Nightly / 2.1.1 Stable

### Operating System (OS)

Linux

### OS Version

Bazzite-Nvidia

### Additional system information

Ryzen 3700X, 16 GB RAM, Nvidia 3070

### Printer

Bambu Labs P1S

### How to reproduce

Open Orca Slicer (had to bypass setup wizard due to blank screen by copying over config files from WIN 10 install).
Navigate to Device Tab.
Press "Play" icon to start camera feed.
Next click (or couple clicks) will crash the program.

### Actual results

OrcaSlicer crashes.

### Expected results

Camera feed should kick on and display video.

### Project file & Debug log uploads

[warning]	2024-08-27 16:49:21.376828[Thread 0x00007fe766f65680]:get_network_function, can not find function bambu_network_del_subscribe
[warning]	2024-08-27 16:49:21.376954[Thread 0x00007fe766f65680]:get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2024-08-27 16:49:21.376967[Thread 0x00007fe766f65680]:get_network_function, can not find function bambu_network_get_profile_3mf
[error]	2024-08-27 16:49:24.182400[Thread 0x00007fe766f65680]:Unable to open the fileG:\Input\RagingHeroes\STL_Modular Parts_UNSUPPORTED\Hole-02.stl
[error]	2024-08-27 16:49:24.182473[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\Charge+holder+progettono+ams.3mf
[error]	2024-08-27 16:49:24.182480[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\Spikey+wall+(revised+tolerances).3mf
[error]	2024-08-27 16:49:24.182487[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\Paint_Handle_1_Piece_15-25mm_A (1).stl
[error]	2024-08-27 16:49:24.182493[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\airbrushstand_hexdesign_nologo.stl
[error]	2024-08-27 16:49:24.182500[Thread 0x00007fe766f65680]:Unable to open the fileG:\Input\00_Blank Bases Round 2_0\Unssuported (FDM print)\TXFA_50mm.stl
[error]	2024-08-27 16:49:24.182506[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\gear_removal_tool.stl
[error]	2024-08-27 16:49:24.182512[Thread 0x00007fe766f65680]:Unable to open the fileG:\Input\Multiboard\Attachments\Army Painter 25.2mm\AP_4Row.step
[error]	2024-08-27 16:49:24.182519[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\32.25mm_4Row.step
[error]	2024-08-27 16:49:24.182528[Thread 0x00007fe766f65680]:Unable to open the fileG:\Input\Pinecil\Pinecil Inserts V2\Pinecil Bottom Large.3mf
[error]	2024-08-27 16:49:24.182534[Thread 0x00007fe766f65680]:Unable to open the fileG:\Input\Pinecil\Peripherals V2\Latch.3mf
[error]	2024-08-27 16:49:24.182540[Thread 0x00007fe766f65680]:Unable to open the fileG:\Input\Pinecil\Outer Case V2\Case Large.3mf
[error]	2024-08-27 16:49:24.182546[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\ptfe_connector_extension_adapter_bambu_print.3mf
[error]	2024-08-27 16:49:24.182553[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\Two sided brush holder tray.3mf
[error]	2024-08-27 16:49:24.182559[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\transparent+window+point+6+mm+nozzle.3mf
[error]	2024-08-27 16:49:24.182565[Thread 0x00007fe766f65680]:Unable to open the fileD:\Downloads\thermometer-magnetic-lower.step
[error]	2024-08-27 16:49:24.182571[Thread 0x00007fe766f65680]:Unable to open the fileG:\Input\RagingHeroes\STL_Modular Parts_UNSUPPORTED\Hole-01.stl
[error]	2024-08-27 16:49:24.182594[Thread 0x00007fe6d8e006c0]:Unable to open the fileG:\Input\RagingHeroes\STL_Modular Parts_UNSUPPORTED\Hole-01.3mf
[warning]	2024-08-27 16:49:24.600026[Thread 0x00007fe766f65680]:04:49:24 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-08-27 16:49:24.619963[Thread 0x00007fe766f65680]:04:49:24 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-08-27 16:49:24.621901[Thread 0x00007fe766f65680]:04:49:24 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-08-27 16:49:24.633090[Thread 0x00007fe766f65680]:04:49:24 PM: Warning: Error running JavaScript: Unsupported result type

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Please note that the errors regarding being unable to find previous projects is fine - due to the nature of having to copy my configs over from WIN 10 just to get past the blank setup wizard screen it's looking for those and can't find them. The camera feed does work great in Bambu Studio, but I prefer the finer control I have access to in Orca (especially when printing ASA).

SuddenGunter, 2024-09-06:
Can confirm the same on  2.1.1 stable, P1s, Win11
When trying to play video feed from printer slicer freezes and the only thing you can do is just kill the process

InfoSec812, 2024-11-11:
I'm getting a similar result on Fedora 40.

```
(orca-slicer:1036232): GLib-GObject-CRITICAL **: 21:01:14.367: invalid cast from 'wxPizza' to 'GtkCellLayout'                                                                                                     
                                                                                                                                                                                                                  
(orca-slicer:1036232): Gtk-CRITICAL **: 21:01:14.367: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed                                                                              
                                                                                                                                                                                                                  
(orca-slicer:1036232): GLib-GObject-CRITICAL **: 21:01:15.301: invalid cast from 'wxPizza' to 'GtkCellLayout'                                                                                                     
                                                                                                                                                                                                                  
(orca-slicer:1036232): Gtk-CRITICAL **: 21:01:15.301: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed                                                                              
                                                                                                                                                                                                                  
(orca-slicer:1036232): GLib-GObject-CRITICAL **: 21:01:15.308: invalid cast from 'wxPizza' to 'GtkCellLayout'                                                                                                     
                                                                                                                                                                                                                  
(orca-slicer:1036232): Gtk-CRITICAL **: 21:01:15.308: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed                                                                              
                                                                                                                                                                                                                  
(orca-slicer:1036232): GLib-GObject-CRITICAL **: 21:01:15.315: invalid cast from 'wxPizza' to 'GtkCellLayout'                                                                                                     
                                                                                                                                                                                                                  
(orca-slicer:1036232): Gtk-CRITICAL **: 21:01:15.315: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed                                                                              
                                                                                                                                                                                                                  
(orca-slicer:1036232): GLib-GObject-CRITICAL **: 21:01:15.320: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:1036232): Gtk-CRITICAL **: 21:01:15.320: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1036232): GLib-GObject-CRITICAL **: 21:01:15.324: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:1036232): Gtk-CRITICAL **: 21:01:15.324: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
BambuTunnel::GetMsg(1)
zsh: segmentation fault (core dumped)  ./Applications/OrcaSlicer_Linux_V2.2.0.AppImage
```

Clicking on the camera feed causes a hang for about 15 seconds and then the application crashes... I tried to enable core dumps, but that doesn't seem to work with AppImages???

Holzniggl, 2024-11-27:
Same on Nobara 40 with Appimage 2.2.0

`(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.977: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.977: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.981: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.981: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.981: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.982: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.982: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.982: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.984: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.985: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.985: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.986: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.986: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.986: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:38:39.986: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:16149): CRITICAL **: 16:38:50.061: Cannot register URI scheme wxfs more than once

** (orca-slicer:16149): CRITICAL **: 16:38:50.061: Cannot register URI scheme memory more than once

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.489: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.489: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.489: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.490: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.491: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.491: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.491: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.491: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.491: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.493: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.493: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.493: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.494: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.494: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.504: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.504: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16149): Gtk-CRITICAL **: 16:40:33.505: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[... 132 lines cut ...]
(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:55.815: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:55.865: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:55.915: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.216: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.266: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.317: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.367: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.417: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.467: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.517: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.567: gst_video_center_rect: assertion 'src->h != 0' failed

(orca-slicer:16149): GStreamer-Video-CRITICAL **: 16:40:56.617: gst_video_center_rect: assertion 'src->h != 0' failed
BambuTunnel::GetMsg(1)
`

Holzniggl, 2024-11-27:
Same Problem on Nobara 40 Appimage 2.2.0

tlhintoq, 2024-12-11:
2024.12.11
P1S just got a firmware update
Orca 2.3.0-dev nightly from 2 nights ago

Still does it.  

- Click the Play button in the UI
- Watch for a bit
- Click stop

Crashes most times.

- *sometimes* if you are very careful to not move the mouse after clicking the `Stop` button and just wait there a few moments it won't crash.
- virtually guaranteed to crash if you play, stop then immediately try to change tabs.

haraldschioberg, 2024-12-19:
Crashes for me immediately when clicking "play" on the camera. Tested with 2.2 and today's 2.3-dev. 

Debian trixie, wayland, AMD graphics. 

```
[....]
(orca-slicer:35922): GLib-GObject-CRITICAL **: 18:00:33.526: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:35922): Gtk-CRITICAL **: 18:00:33.526: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:35922): GLib-GObject-CRITICAL **: 18:00:34.324: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:35922): Gtk-CRITICAL **: 18:00:34.324: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
BambuTunnel::GetMsg(1)
Segmentation fault
```

Ruthalas, 2025-01-05:
I am also experiencing this issue, and haven't had any success with the work-around suggested by @tlhintoq .
I see several issues that _may_ be related, some of which have been closed as stale: #2439 #5878 (#6431 #2896).

#5878 has a [comment](https://github.com/SoftFever/OrcaSlicer/issues/5878#issuecomment-2480347113) from @ronoaldo in which he speculates on some possible causes.

I'd love for this to be fixed because this is the single thing preventing me from swapping off on BambuStudio.
The time you dedicate to this tool is much appreciated.

Ruthalas, 2025-04-12:
This issue continues to persist. Do not close.
(Three of the four other referenced issues have been closed as stale with no solution.)

dmayhem, 2025-07-04:
This issue has been around for at least a year and a half. And has been reported before.
Seems to be specific to Fedora and it's variants? Also seems to be related to gstreamer. I don't think I've seen any reports on Ubuntu and it's variants.
Still definitely present in Fedora 42, with a fresh build from the nightly-builds tag

dmayhem, 2025-07-05:
Tried with v2.3.0 appimage
Tried with fresh compile of 2.3.1 nightly (both with and without appimage)

Both crashed, same as before
Fedora 42 KDE Wayland amdgpu video

ryodeushii, 2025-07-05:
> Tried with v2.3.0 appimage Tried with fresh compile of 2.3.1 nightly (both with and without appimage)
> 
> Both crashed, same as before Fedora 42 KDE Wayland amdgpu video

what is in dmesg after crash?

dmayhem, 2025-07-05:
Nothing in dmesg.
console output from OrcaSlicer:
```
BambuTunnel::GetMsg(1)
Segmentation fault (core dumped)
```
Probably going to rebuild Orca with debug info when I get some time, and run it via gdb.

ryodeushii, 2025-08-01:
> I can second the crash still happening, despite adding the ENV.

Here's my desktop file (`~/.local/share/applications/OrcaSlicerFixed.desktop`) just in case it might be helpful.
```
[Desktop Entry]
Name=OrcaSlicer - Fixed
GenericName=3D Printing Software
Icon=OrcaSlicer
Exec=env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 GBM_BACKEND=dri WEBKIT_SKIA_ENABLE_CPU_RENDERING=1 /usr/bin/orca-slicer %u
# Exec=orca-slicer %U
Terminal=false
Type=Application
MimeType=model/stl;model/3mf;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;x-scheme-handler/orcaslicer;
Categories=Graphics;3DGraphics;Engineering;
Keywords=3D;Printing;Slicer;slice;3D;printer;convert;gcode;stl;obj;amf;SLA
StartupNotify=false
StartupWMClass=orca-slicer
```

ChristianJohnsten, 2025-08-30:
I am on Fedora 41 Plasma, and have tried using the `WEBKIT_SKIA_ENABLE_CPU_RENDERING=1` environment variable and it still crashes when I try to access my Bambu Lab P1S camera

KyleKirkpatrick, 2025-10-03:
Also have been experiencing this bug on 2.3.1-beta (build 1e87666) using Windows and Bambu P1P. 

Successfully worked around the bug by forcing Windows to choose the integrated GPU instead of the discrete GPU (per @cianhorgan 's suggestion). This work-around is unfortunate though, as I would like to use the dGPU to display complex models in Orca Slicer. 

Orca 2.3.1-beta (build 1e87666)
Operating System: Windows 11 Pro 64-bit (10.0, Build 26100) (26100.ge_release.240331-1435)
System Manufacturer: Microsoft Corporation
System Model: Surface Book 3
Processor: Intel(R) Core(TM) i7-1065G7 CPU @ 1.30GHz (8 CPUs), ~1.5GHz
iGPU: Intel(R) Iris(R) Plus Graphics
dGPU: NVIDIA GeForce GTX 1660 Ti with Max-Q Design

liweitianux, 2025-11-17:
I'm having the same crash with the latest 2.3.1 AppImage on Debian forky/testing with P1S. I'm using Orca on ThinkPad X1 Carbon that only has an iGPU.