OrcaSlicer/OrcaSlicer #6598: If 3d Mouse (Spacemouse) connected then ignore 2d mouse for world pan, tilt, zoom, move etc.

tlhintoq, 2024-08-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is so, so, so easy to miss the model or the various control arrows, zoom nodes, rotate handles etc...
If I already have a 3d mouse such as a Connexion Spacement then let me ignore the 2 mouse for all that stuff because I'm going to use the 3d mouse for environment movement anyway.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Ideally - if there is a 3d mouse just ignore world adjustments from the 2d mouse.  Leave the 2d for grabbing objects and controls and so on.

### Describe alternatives you've considered

Maybe a Globe&lock icon on the front UI (with an easy hot key that we can program on the 3d mouse anyway)... if that's locked then the world is locked in place, making it easier to manipulate objects without constantly panning the entire world by accident.  

### Additional context

[image]

foreachthing, 2024-08-30:
> Ideally - if there is a 3d mouse just ignore world adjustments from the 2d mouse. Leave the 2d for grabbing objects and controls and so on.

No, please don't just force something onto other users. I, too, have a SpaceMouse and still use the mouse sometimes - not much. If anything, make it optional. I wish one could customize all navigation controls...

tlhintoq, 2024-08-30:
Yeah a checkbox for `[x] Ignore 2d world navigation` or however to phrase it sounds good

closed 2024-12-06 by github-actions[bot]