OrcaSlicer/OrcaSlicer #6600: Create a Docker Compose setup to optimize and speed up compile times

longrackslabs, 2024-08-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I noticed this comment in DockerBuild.sh :

# Wishlist hint:  For developers, creating a Docker Compose 
# setup with persistent volumes for the build & deps directories
# would speed up recompile times significantly.  For end users,
# the simplicity of a single Docker image and a one-time compilation
# seems better.

So decided to give it a go. doing testing now.  Not sure how issues  get assigned, but i should have a PR shortly.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Create a Docker Compose setup to optimize docker image creation, rebuilds, etc.  Work should consist of:

1. docker-compose.yml
2. ComposeDocker.sh
3. Testing that above generates a correct docker container and runnable application.  Existing scripts such as BuildLinux.sh, DockerBuild.sh, DockerRun.sh should not be effected and should all still work.

### Describe alternatives you've considered

This seems like a straight forward and standard solution to using docker-compose to create a docker image.

TrentBartlem, 2024-08-31:
Could probably put up a pull request and say 'fixes https://github.com/SoftFever/OrcaSlicer/issues/6600' in the commit message?

longrackslabs, 2024-08-31:
Yeah, I'm working on that. I've got the docker compose working. Just need to work on optimizing the build.

longrackslabs, 2024-09-05:
I have been working on this but am having problems with rebuilds or partial rebuilds, like -d to rebuild dependencies.  it errors after the last link step.

closed 2024-12-13 by github-actions[bot]