OrcaSlicer/OrcaSlicer #6601: Allow to slice even thoght colisions may ocur

alc112, 2024-08-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I wouldn't say it's a problem. The way this is set now is safe for beginners.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

When I want to print something in "Print Sequence by Object" mode and Orca detects a potential collision, it simply won't let me slice the project even if I set Extruder Clearance radius to zero. 

### Describe alternatives you've considered

If I'm fully aware of the risk and I know that nothing will happen, I'd like a way to bypass this limitation. Perhaps a pop-up warning where Orca explains the potential risks and allows me to proceed if I choose to.

### Additional context

I'm making keychains of 2 layers thick. This thickness is not enough to cause collisions. In fact, I do this all the time in other slicers but Orca just won't let me continue.

bo0tzz, 2024-11-28:
I don't think this is stale

MAJCcustoms, 2025-02-19:
I'm not sure why there isn't an option to override it or disable collision detection altogether. I've already tried lowering the extruder clearance to 0 and it still doesn't let me slice

loutwice, 2025-02-28:
@alc112  i think we got the same idea ,i just made my ticket https://github.com/SoftFever/OrcaSlicer/issues/8608