OrcaSlicer/OrcaSlicer #6602: If there's no change in bed temp, don't send command to change the bed temp

tlhintoq, 2024-08-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-dev

### Operating System (OS)

Windows

### OS Version

Win10x64

### Additional system information

[image]

### Printer

Elegoo giga

### How to reproduce

First you have to understand that the Giga has 4 heatbeds.  If the part is only on bed T2 you don't need to heat the other 3.
Its an easy matter to make a change to only warm the used bed by making the bed command specify the right bed for that job.
[image]

HOWEVER... Orca is generating a change to the bed temp after the first layer even if the temp is the same.
From: 65 to 65 for example.  This command does not appear in any material gCode to be overwritten.
So an Giga user can't just `Slice` and `Print`.  They have to slice, save to usb, go to the machine on the shop floor, change USB drives and launch from the machine.

[image]

**How to reproduce:** Slice any job where the bed temp is unchanged between first and following layers.

### Actual results

After the first layer the next is dynamically injected commands that we can't stop or change.
[image]

### Expected results

Don't issue changes that aren't needed.  From 65 to 65 doesn't require another command be sent.

### Project file & Debug log uploads

None required

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

tlhintoq, 2024-09-22:
FYI: Elegoo's slicer has fixed this bug - so I don't think its all that hard.
Can someone out there just do a compare against the two branches and reverse-merge their fix so Orca can be better used by Giga owners?  Please!
cc: @Noisyfox

Noisyfox, 2024-09-23:
> Elegoo's slicer has fixed this bug

Do you mean their Cura-based slicer?

tlhintoq, 2024-09-23:
> > Elegoo's slicer has fixed this bug
> 
> Do you mean their Cura-based slicer?

No - their OrcaBased fork that they are branding in their own name. - I'm not happy either.  They should have just been a contributor here and maintained being a part of the open-source mind set.  Then again - they needed to deploy a slicer that didn't have this bug.

Noisyfox, 2024-09-23:
> No - their OrcaBased fork that they are branding in their own name. - I'm not happy either. They should have just been a contributor here and maintained being a part of the open-source mind set. [...]

Did they publish the source code of that anywhere? I couldn't find it.

tlhintoq, 2024-09-23:
> Did they publish the source code of that anywhere? I couldn't find it.

https://github.com/ELEGOOTechSupport/ElegooSlicer

d-portero, 2024-09-26:
this would be very helpful!

ELEGOOTechSupport, 2024-10-22:
[image]

ELEGOOTechSupport, 2024-10-22:
[image]

https://github.com/ELEGOO-Slicer/OrcaSlicer_PR/commits/refs/heads/Partition-hot-bed-gcode-processing/

[image]

It is optimized in this PR.

closed 2025-01-28 by github-actions[bot]