OrcaSlicer/OrcaSlicer #6603: Ability to change colors of pre-painted 3mf

Doogle03, 2024-08-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Pre-painted 3mf files have pre-set tool numbers.  Would like ability to change which tool is applied to painted areas. (everything painted white, I now want painted black, but black is already loaded on tool 4 and file is painted with tool 1)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Would like the ability to change which tools are applied to pre-painted 3mf files.  

Eg. File comes pre-painted with 3 colors, selecting tool 1 through tool 3, but want to use the colors on other tools.  Have option to change all portions painted with tool 'x' and replace with tool 'y' without needing to repaint or attempting to complete a manual recolouring with the fill tool in slicer.

### Describe alternatives you've considered

Manually trying to re-paint file could result in some areas not properly being changed, fill tool does not catch all portions of previous color.  It is very tedious and prone to error

### Additional context

[image]

Pre-painted file, want to change colors to different tools via drop-down box.  Change all white to black, change all yellow to white, change all pink to brown

adrianbear, 2024-08-30:
I'd like to give this a +1 please

closed 2024-12-06 by github-actions[bot]

buergi, 2026-02-09:
I think this issue is already implemented. In color painting mode click "Remap filaments" to re-map the filament colors

[image]

Has been introduced by https://github.com/OrcaSlicer/OrcaSlicer/pull/10303