OrcaSlicer/OrcaSlicer #6604: Pressure calibration doesn't work

Idaho947, 2024-08-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

AMD Ryzen 7 3700U with Radeon Vega Mobile Gfx     2.30 GHz
20,0 Go RAM

### Printer

Creality ender 3 S1 pro

### How to reproduce

Go to calibration
Go to pressure advance
Select tour
Clique sur OK

### Actual results

Two message :
The template file failed to load.

And :
The file does not contain any geometric data.

Orca crash

I saw that there could be a problem with line height but nothing at me it is 0.45

### Expected results

The tower is created

### Project file & Debug log uploads

[crash_Fri_Aug_30_08_19_49_0.log](https://github.com/user-attachments/files/16812615/crash_Fri_Aug_30_08_19_49_0.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

No project files because orca crash before

discip, 2024-08-30:
@Idaho947
Have you tried to slice it?

Idaho947, 2024-08-30:
Nop I can't it bult noting on the bed. I've this error when I click on OK

Bastlwastl84, 2024-10-05:
same here with error Message "Verarbeitung der Vorlage filename_format fehlgeschlagen.
Parsing error at line 1: Non-integer index is not allowed to address a vector variable.
{input_filename_base}_{filament_type[initial_tool]}_{print_time}.gcode
                                     ^"
Kalibrierung>Pressure Advance>PA-Muster not working

Bastlwastl84, 2024-10-05:
[image]
selecting PA Muster

[image]
the typical cube apears

[image]
the PA-gcode is not calculated correctly

[image]
"Aktuelle Platte drucken" is grey now

[image]
changing to "Aktuelle Platte drucken"

[image]
if pressing "Aktuelle Platte druckem" the Error Message pop up

[image]

closed 2025-01-11 by github-actions[bot]