OrcaSlicer/OrcaSlicer #6612: 1 out of 2 starts the configuration does not appear

Elrohir80, 2024-08-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

up tu 2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Two trees SK1

### How to reproduce

The first time I open Orca Slicer, it doesn't look like filament or seting settings, but the printers I have added do appear. I close and enter again and if everything appears, I close and open again and it does not appear. Happens in odd starts
[image]
[image]
[image]
[image]

### Actual results

When I start the second or fourth time, it works. I HAVE added orca as an admin and to the windows firewall but it does not solve it

### Expected results

This has been happening to me since version 2

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/16821335/logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

traxman25, 2024-09-04:
Same issue here also.

Elrohir80, 2024-09-04:
> Same issue here also.

thanks. I was starting to think that it was only happening to me

traxman25, 2024-12-04:
Issue still persists, it is not a dead issue.

Elrohir80, 2024-12-04:
Good day
Yesterday I managed to "solve" the problem by creating a printer and using the original profile values. It was tedious since I had to create the printing and filament profiles again, but now at least I only have to open Orca once.
As I said, this didn't happen to me at first, it was after the upgrade to version 2x

[image]
[image]
[image]

closed 2025-03-13 by github-actions[bot]