OrcaSlicer/OrcaSlicer #6615: Color sequence not respected when print sequence is by object

andreaselias, 2024-08-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

Sonoma 14.4.1

### Printer

Bambu A1

### How to reproduce

Add 2 different models to the plate, one with multiple colors.
On the plate settings set the print sequence by object
Select the color sequence

### Actual results

First model that has a solid color prints as expected
Second model print with a color different from the one selected

On the attached image we can see that the green color is printed first then the yellow and red last even though I set to print the red first

[image]

### Expected results

Print the defined sequence as bambu studio does

[image]

### Project file & Debug log uploads

[debug_Fri_Aug_30_19_02_00_9758.log.0.zip](https://github.com/user-attachments/files/16822593/debug_Fri_Aug_30_19_02_00_9758.log.0.zip)
[Mathematicus little 1x1.3mf.zip](https://github.com/user-attachments/files/16822595/Mathematicus.little.1x1.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2024-09-06:
Could you provice the 3mf from BBS as well?

Noisyfox, 2024-09-06:
[image]
As you can see, BBS has the same issue as Orca unfortunately.

andreaselias, 2024-09-06:
Well I thought I was going crazy but I found why it showed the correct print sequence in my original post. If I set the correct order in the filament section it slices correctly. So it looks like that the color order is been ignored when the print sequence is by object.

Wrong order in the filament section
[image]

Correct order in the filament section
[image]

Noisyfox, 2024-09-06:
Could you try the PR build and see if that solves your problem?

chrpoe, 2024-12-22:
Hey there,
I was searching and trying and testing - and I can confirm:
When "Print by Object", rather than "Print by Layer", the "First Layer Filament Sequence" is ignored.

The filament sequence for other layers works as specified.

So the bug only happens for:
/ Print by Object
/ First Layer Filament Sequence

closed 2025-01-22 by Noisyfox