OrcaSlicer/OrcaSlicer #6618: Tree Support - Hybrid not supporting flat overhangs properly?

Arakon, 2024-08-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

Bambu A1

### How to reproduce

- Enable Tree Supports Hybrid
- Slice a model with 90° overhangs
- The result is a barely supported surface

### Actual results

The overhang is barely supported, with lots of tiny support tips that are highly likely to get torn off when bridging.
[image]

### Expected results

Similar to Tree Organic, a full and supported top interface should be created between layer and supports:
[image]

### Project file & Debug log uploads

[treesupport.zip](https://github.com/user-attachments/files/16824056/treesupport.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ichisaur, 2024-09-05:
I can confirm that this weirdly deviates from Bambu Studio.

Hybrid support in Bambu Studio produces a standard style support for the flat area and square tree supports with "On Build Plate Only" Disabled
[image]
[image]

In Orcaslicer, with "On Build Plate Only" enabled, you get a forest of tree branches. 
[image]

In Orca, with "On Build Plate Only" Disabled, it is a more standard grid, but no square trees, and standard grid for levels. 
[image]

The "On Build Plate Only" behavior is not replicated in Bambu Studio.

closed 2024-12-13 by github-actions[bot]

roxorplox, 2025-07-21:
This needs to be re-opened. Hybrid Tree supports arent functioning properly.