OrcaSlicer/OrcaSlicer #6620: Gap infill with multiple filaments being done with wrong filament

ryebreadsalad, 2024-08-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-dev

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5

### Printer

Bambu P1P/S

### How to reproduce

1. Have a model/box and some text created in CAD. Text is 0.25mm in height (first layer height), as is the indentation in the model.
2. Choose a second filament for the text
3. allow gap infill for the text and don't filter out the small areas in my case
4. Slice.
Issue does not appear when only slicing the text without other model.
Issue persists even when changing filament order.
Issue does not appear in official version 2.1.1
Using the same project file.

### Actual results

 Gap Infill for the text is being done in the wrong filament which in my case is all of the center parts of the text which makes the text dual color.

[image]

### Expected results

The text as well as its gap infill should be printed with the selected filament for the text.

[image]

### Project file & Debug log uploads

[OrcaColorBug v1.3mf.zip](https://github.com/user-attachments/files/16824846/OrcaColorBug.v1.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2024-09-05:
The problem is this part:
https://github.com/SoftFever/OrcaSlicer/blob/c179a577250323eb9f7862d73d77b7f20536aaac/src/libslic3r/GCode/ToolOrdering.cpp#L178-L188

Which causes the gap fill to always be printed with extruder 0.

SoftFever, 2024-09-08:
@ryebreadsalad 
Thanks for reporting it. Nice catch!

@Noisyfox Thanks for the triage, which saved a ton of time digging through for me!👍 
The bug was introduced when I rewrote this part while adding support for the multi-tool.

It's now fixed.

PR #6680 (2024-09-08): Bugfix: gapfill was assigned with wrong extruder/filament
# Description

Fixed #6620
commit 11240b4a (2024-09-08): Bugfix: gapfill was assigned with wrong extruder/filament (#6680)

# Description

Fixed #6620
closed 2024-09-08 by SoftFever